cortex-lab/phy #94: Better algorithm for auto WaveformView display

nippoo, 2015-02-07:
At the moment, the waveforms are just positioned based on their probe location; it is possible to pan and zoom in both axes to refine this, as well as increase/decrease amplitude but this should be automatic. Default view for IMEC-128ch is not ideal.

i.e. minimize whitespace in the view: probably something along the lines of

- set waveform width so that the total width takes up 80% of the view (maybe set a maximum default width for each waveform, too wide looks odd)
- scale so the whitespace is evenly distributed
- set waveform amplitude so that they don't overlap (maybe take ~5 spikes from each cluster when calculating this)

rossant, 2015-02-07:
Related to #54

rossant, 2015-04-04:
I think we should not be too smart with this. It should be enough to:

* Document properly the keyboard shortcuts to update the view, change the amplitude of the waveforms, etc.
* Automatically save all parameters upon closing on a per-dataset basis (see #139).

nippoo, 2015-10-04:
Reopening and setting milestone to 0.3. Part of GUI improvements.

closed 2016-03-26 by rossant