cortex-lab/phy #105: Programmatic high-contrast colour generation

nippoo, 2015-02-11:
- Generate a list of colours automatically based on highest-contrast between neighbouring colours (see http://devmag.org.za/2012/07/29/how-to-choose-colours-procedurally-algorithms/).
- Return white/black highest contrast text (several algorithms to do this!)
- Needs to be colourblind-friendly
- Can specify "must be bright" or "must be dark" or "must be neutral" - high-contrast with black or white, or both

rossant, 2015-02-11:
see https://github.com/kwikteam/phy/blob/master/phy/utils/_color.py#L26

rossant, 2015-02-11:
Maybe add:

* Generate manually a colormap of 10-20 (even 30?) colors that are:
    * light enough to be well visible on a black background
    * color n should be quite distinct from color n+1

We'll use this colormap when assigning colors to new clusters.

rossant, 2015-03-04:
@nippoo It might be sufficient to generate a single colormap that has all required properties. The first selected cluster will get the first color, the second one the second color, etc.

rossant, 2015-04-10:
There now is a good enough colormap

closed 2015-04-10 by rossant