cortex-lab/phy #114: ClusterStore

rossant, 2015-02-19:
In-memory structure storing per-cluster information that only depends on the recording data (i.e. not user-dependent information like cluster group, which comes with its own undo stack)

Closes #58, closes #70, closes #99.

* Some of this information may be memory-mapped (like the list of all features and masks per cluster), using an on-disk data store

* Every piece of information can be in a MemoryStore or a DiskStore. This is transparent in the external API. Every field choses how it's going to be stored.

* Modular architecture: it is easy to add/modify the kind of information stored in the structure

* Every cluster is identified by an `id`, which is a non-negative integer

* Fundamental hypothesis: the information of cluster `<id>` cannot change during a clustering session. New clusters can be created, and clusters can be deleted, but no cluster is modified. Merging/splitting => delete/new clusters.

* TODO: currently, `Clustering.assign()` *can* result in clusters being modified without their IDs being changed (for example, split). This need to change!

* This structure is completely decoupled from the file format. There are import/export functions from/to the file format.

* There is an `update(up)` method that is called when clustering assignements change (merge/split/assign). The structure should update accordingly.

* Initial list of fields in the structure; for each cluster:
  * n_spikes (memory)
  * spike_ids (memory)
  * features (disk)
  * masks (disk)
  * waveforms (raw data on disk)
  * mean_masks (memory)
  * mean_waveforms (memory)
  * quality (memory)  # find a more specific name
  * n_unmasked_channels (memory)
  * mean_probe_position (memory)

* Rename `spike_label` and `cluster_label` to `spike_id` and `cluster_id`

* There are several events:
  * `import`: when data is initially imported and all cached data is cleared
  * `export`: when internal data is exported to a public file
  * `update`: when clusters change

* There are several methods:
  * `_store(cluster, ...)`
  * `_load(cluster, ...)`
  * `delete(clusters)`

* Every field can subscribe to any event:

  ```python
  @cluster_store.connect
  def on_import(model, cluster):
      """Called when data is imported, for every cluster."""
      # get my_features for the given cluster from the model
      self._store(name='features', data=my_features, cluster=cluster,
                  storage='disk')

  @cluster_store.connect
  def on_import(model):
      """Called when data is imported, just once."""
      self._store(..., storage='memory')

  @cluster_store.connect
  def on_update(up):
      if up['description'] == 'merge':
          self._store(...)
          self._delete(...)

  @cluster_store.connect
  def waveforms(cluster):
    self._load(name='waveforms', cluster=cluster)

  @cluster_store.connect
  def mean_masks(cluster):
    self._load(name='mean_masks', cluster=cluster)
  ```

* For disk storage, we need a private class handling file management (one file per cluster):

  ```python
  # Create the folder if it doesn't exist
  ds = DiskStore(some_folder)

  ds.store(name='features', data=my_features, cluster=cluster)
  ds.load(name='features', cluster=cluster)
  ds.delete(clusters)

  ```

* Idem for MemoryStore (very simple: just wrapping a dict).

* TODO: progress bar API (when initially importing a dataset)

ping @nippoo @ctaf42

ctaf42, 2015-02-19:
looks good.
Why do we need event for the internal logic?
It is better to group loading (from the same source) in one single fonction. (reduce read, reduce cb overhead, ...)

rossant, 2015-02-19:
> Why do we need event for the internal logic?

It offers a modular interface to specify the fields we want in the store. If we want to add or modify some information, we have just one place to update, not 10 like in the current program. I use the same approach in the other parts of the library (see https://github.com/kwikteam/phy/blob/master/phy/cluster/manual/interface.py for example). The implementation is very easy: the entire system is just a few lines of Python.

> It is better to group loading (from the same source) in one single fonction. (reduce read, reduce cb overhead, ...)

When implementing the system I'll make sure to reduce the overhead. But importing a dataset will be a heavy operation anyway, to be performed only the first time you load a given dataset. I think it's okay if it takes 1 minute or so (to read a multi-GB file, loading and rearranging the data, and writing hundreds of smaller filer).

PR #116 (2015-02-27): Cluster store
This PR brings significant refactoring changes and new features. The clustering logic has been strengthened, and there is now a ClusterStore that holds per-cluster information in memory and on disk.

* [x] MemoryStore
* [x] DiskStore
* [x] Generic event emitter, Session derives from it
* [x] BaseClusterStore with event system
* [x] _all_spikes_in_clusters()
* [x] Clustering now has an in-sync spikes_per_cluster structure
* [x] ProgressReporter
* [x] Rename `label` to `id` for clusters and spikes
* [x] Update cluster assign such that modified clusters get a new id
* [x] Transform  [... cut]
closed 2015-02-27 by nippoo