cortex-lab/phy #115: Test minimal no-anaconda environment on Travis

rossant, 2015-02-20:
With `apt-get` exclusively.

rossant, 2015-02-20:
See [this](https://github.com/vispy/vispy/blob/master/.travis.yml) for an example of a `.travis.yml` with multiple configs.

nippoo, 2015-02-20:
What should our minimal supported config be?

rossant, 2015-02-20:
I would suggest to try the latest Ubuntu LTS (so 14.04) with the latest Ubuntu apt-get packages for NumPy, h5py, etc. We'll see if we get incompatibilities then.

nippoo, 2015-02-20:
Okay! I suspect we'll have to install IPython manually with pip though?... since the stable version is 1.2 on Ubuntu's repo

nippoo, 2015-02-20:
Do we want it to be Python2 or Python3? Or we could do both.

rossant, 2015-02-20:
Hmm, yes probably. Hopefully that should be fine because IPython is pure Python. But the notebook comes with many dependencies. (PS: IPython 3.0 will be released in a couple of weeks)

rossant, 2015-02-20:
> Do we want it to be Python2 or Python3? Or we could do both.

I would say Python 2 only for now. We can assume that those who want bleeding edge Python 3 will be okay with using anaconda...

closed 2016-03-24 by rossant