cortex-lab/phy #118: Consider using git for the undo stack

rossant, 2015-02-27:
Currently, cluster changes are saved in a Python list: `(spike_ids, cluster_ids)`. Undoing = replaying the entire past sequence except the last item. This is lost when closing the session.

Another option would be to save the entire `spike_clusters` array in a text file, with one line per spike, containing the cluster id for that spike (this is the old `.clu` file). This text file would be saved in a user temporary directory, and it would be version-controlled with git. It may be part of the cluster store. The user would never have to see this.

* Location: `~/.phy/my_file/clustering/0/main/spike_clusters.txt` and `cluster_groups.txt` (0 is the shank)
* Every clustering change (merge, split, etc.) = 1 commit
* Undo = go back in the branch
* How to deal with branching? Example: merge, merge, undo, split. How does git deal with that?
* cluster_groups can contain `3 noise` or `7 good` on each line, with the cluster_id and group name on every line.
* This is saved into the kwik file only when closing the dataset (or when manually clicking on the "Save" button?). Saving = copying the contents of these files into the kwik file.

**Pros**

* **Never loose your data again**. Every action is saved and reasonably safe from data corruption. If the computer crashes, everything has been automatically saved.
* Performance?
* Unlimited undo stack, even across multiple sessions.
* Does not take any RAM (but does take disk space).

**Cons**

* Hard dependency on git, harder to distribute, Windows users, etc.

Ping @nippoo @ctaf42

ctaf42, 2015-02-27:
What does git brings, that simple texts files doesn't if we are speaking of temporary files not shared?

IE:
do we need compression? we may. but its very to do file by file too.
do we need to exchange the history with others or share it accros boxes?
do we need to merge histories together?

rossant, 2015-02-27:
History!

ctaf42, 2015-02-27:
flat history is easy with files:
h.1
h.2
h.3

tree is history is a little bit harder, but still very easy to do =>
h.hash1
h.hash2
h.hash3
h.index
h.index define the tree that refer to all temporary files.

Very simple and straightforward to implement, without the complexity of git, nor the dependency.

rossant, 2015-02-27:
That's a possibility indeed. But the point with using git was that it was already implemented. If we have to implement it ourselves (even if it's simple) we may as well keep things as they are for now.

ctaf42, 2015-02-27:
I'am pretty sure, the implementation for what I propose is shorter than the one using git, and more maintainable. It's a matter of using pickle.

you dont need the feature of git at all.  to me using git here feels like using a bazouka to kill a moskito.

rossant, 2015-02-27:
Yeah you may be right. Instead of pickle I would probably use text files though, unless the perf is too bad.

ctaf42, 2015-02-27:
was speaking of pickle in textfile (unless perf are too bad).

for example :
pickle.dump(mynode, open( "file.hash", "wb" ) )
pickle.dump(mytree, open( "tree", "wb" ) )

nippoo, 2015-05-28:
YAGNI

closed 2015-05-28 by nippoo