cortex-lab/phy #156: Ticks and baseline in CCG view

rossant, 2015-04-06:


nsteinme, 2015-04-29:
Would also be good to show a vertical line at some settable value, e.g. 3 ms, so it's easy to see if there are refractory violations. In the future it would be great to also show the results of some calculations related to this on these plots: for ACG, the estimated false positive rate (see Hill et al); for CCG, perhaps what the false positive rate would be if the two units were merged. 

Also, the colors are sort of faded relative to what they look like in the other views?

rossant, 2015-04-29:
+1

About faded colors: can you send a screenshot? This should not be happening.

nsteinme, 2015-04-29:
Here - 
[image]

nsteinme, 2015-04-29:
Not dramatically different. Might get confusing with lots of colors though.

rossant, 2015-04-29:
Where are the faded colors here? They look similar to me indeed

nsteinme, 2015-04-29:
Here are samples from CCG, waveform, and feature, left to right:
[image]

The RGB values are:
102/194/165 in CCG
0/254/174 in waveform and in feature

rossant, 2015-04-29:
Ah but there is some alpha blending in waveform and feature + saturation and value which depend on the masks. I think you would have the same color if you display just 1 spike with mask=1.

In any case is this a problem? Do you have something to suggest otherwise? We can easily tweak the colors in many ways.

nsteinme, 2015-04-29:
It's not a problem as long as the clusters you're looking at are all represented with very different colors. It's not common to want to look at 10 or more clusters simultaneously, but if you did then it would probably get confusing which waveform mapped to which CCG. That's all. No big deal, just pointing it out.

rossant, 2015-04-29:
No problem at all :) I'll see what I can do...

nippoo, 2015-05-28:
Faded colours is fixed now :)

rossant, 2015-06-08:
@nsteinme  OK, everything should be done here. In the upcoming PR you can do something like `correlograms.lines = [-60, 0, 60]` to set three vertical lines at -60, 0, 60 time samples (low-level interface). In the high-level interface what about something like `correlograms.set_refractory_line(3.0)` ?

Also, for the other issue, there is now a statistics panel where you can add your own statistics under the form of a function `mystat(cluster)`. Currently, only the stats of the first selected cluster is shown. Do you need to see the stats of all selected clusters? (see #324)

nsteinme, 2015-06-10:
Hey Cyrille.... sorry, where do I find "correlograms"? It doesn't show up
for me in the default namespace. There is gui.views[4] which is the
correlograms view, but it doesn't have a lines property or method, so far
as I can see.

rossant, 2015-06-10:
this isn't well documented yet -- try `gui.views[4].view.lines`? what is the class of `gui.views[4]`? it should be CorrelogramsViewModel or something...

nsteinme, 2015-06-10:
Great, that worked. Are the units in samples or milliseconds?

rossant, 2015-06-10:
samples!

nsteinme, 2015-06-10:
Well, I know you said that in your example above, but when I set lines to [-5, 0, 5], I see this:

[image]

and I don't think it's possible to have spikes that are just a couple of samples apart. That looks more like 5ms to me, but I wasn't sure. Sampling freq of this recording is 25kHz. 

A different question, above you suggested I can add a statistic of the form mystat(cluster). What is "cluster"? Is there a cluster object? If so, how can I figure out what would be in it?

rossant, 2015-06-10:
sorry, the unit is in number of bins...

rossant, 2015-06-10:
> A different question, above you suggested I can add a statistic of the form mystat(cluster). What is "cluster"? Is there a cluster object? If so, how can I figure out what would be in it?

cluster is just cluster_id, an integer

nsteinme, 2015-06-10:
So then does my function have access to something that would let it know,
e.g. the spike times of that cluster?

nsteinme, 2015-06-10:
Ok, that makes more sense, how do I tell what the bin size is? I don't see
it in gui.views[4].view.?

rossant, 2015-06-11:
`gui.views[4].binsize`: number of samples in every bin
`gui.views[4].winsize_bins`: size of the CCG window in number of bins

rossant, 2015-06-11:
You also have, on the same view model:

```

    def change_bins(self, bin=None, half_width=None):
        """Change the parameters of the correlograms.

        Parameters
        ----------
        bin : float (ms)
            Bin size.
        half_width : float (ms)
            Half window size.

        """
```

rossant, 2015-06-11:
> So then does my function have access to something that would let it know,
> e.g. the spike times of that cluster?

if you're in an interactive session, you have access to `session.model` and `session.store`. I'll work on the case where you put your function in `user_settings.py`

closed 2015-06-16 by nippoo