cortex-lab/phy #190: ISI view

rossant, 2015-04-13:
Code can be shared with the existing CCG view.

rossant, 2015-07-02:
subplot(i, j) = ISI merge i & j

rossant, 2015-08-03:
@nippoo @yger this can be something for one of you..

yger, 2015-08-03:
I ll try to give it a try, modulo understanding the code...

Best

Pierre

nippoo, 2015-10-06:
@yger you want to do this before SfN or shall I see if I have time?

yger, 2015-10-06:
Well... To be fair, I won't have time, I'm afraid: I'm already 
overwhelmed in finishing our poster and some simulations, and packaging 
our software into something decent. I would not make this ISI view on 
the top of the priority list, so this is up to you.
As far as I understood after discussions with you guys, key stuffs to 
see in phy would be: amplitudes and templates superimposed onto the raw 
traces. And, of course, for us to be able to use phy on a daily basis 
and then help more to the code itself, not computing all the features 
for all the spikes... The ISIs view is something useful that we are 
using, but you said you are only using the CCs view

Maybe we can talk about it on Friday, if you are around with Cyrille?

Best

Pierre

rossant, 2019-06-13:
done in the dev branch

closed 2019-06-13 by rossant