cortex-lab/phy #198: Option to only zoom in the subplot hovered by the mouse pointer

rossant, 2015-04-14:
Can use a uniform with the index of the currently-hovered subplot, and if statement for pan/zoom in GLSL.

rossant, 2015-04-14:
@shabnamkadir @nsteinme @nippoo at some point I'd appreciate feedback on the pan/zoom UX in the 3 different views. I've improved it in an upcoming PR, but there are still some minor issues.

rossant, 2015-04-29:
@nsteinme thanks for the feedback. Do you think it would be interesting to have an option to switch between the two behaviors (current one, and independent zooming in each plot), or is it okay to just have the independent zooming option?

nsteinme, 2015-04-29:
I think switching between both is useful - it's good to do all at once if, for instance, they're all wildly mis-scaled at the moment. But you can definitely imagine wanting to scale them independently. 

Btw there was one part of the other issue I posted that isn't covered here (sorry for double-posting again) - should be able to scale y-axis separately from x. I didn't figure out how to do this yet (it doesn't work like in KV, at least).

rossant, 2015-04-29:
> Btw there was one part of the other issue I posted that isn't covered here (sorry for double-posting again) - should be able to scale y-axis separately from x. I didn't figure out how to do this yet (it doesn't work like in KV, at least).

Ah sorry I had missed that. Are you referring specifically to the feature view, or the other views too? I figured it wasn't that useful to scale the axes independently, and it seemed to me that most users were confused about it (or didn't even know that feature!). In many cases you want to keep the aspect ratio fixed. Could you precise in which situation it is useful to scale the axes independently?

nippoo, 2015-04-29:
The default behaviour should definitely be to scale the views
independently. Perhaps holding down a modifier key could trigger rescaling
all the views? In any case, I'm not sure if/why you'd want to pan them all,
it might be limited only to zoom, since they're all projections of
different dimensions. I can definitely seeing it be more of an annoyance if
it's easy to do it accidentally...

nsteinme, 2015-04-29:
I use it all the time in KV when one dimension is much larger than another, e.g. to go from this (the default original appearance):

[image]

to this: 

[image]

nsteinme, 2015-04-29:
In KV you just right click and drag around - things scale in a really nice and intuitive way. This behavior should be retained, I think.

rossant, 2015-04-29:
> In KV you just right click and drag around - things scale in a really nice and intuitive way. This behavior should be retained, I think.

Personally I agree, but you're the first one I know who thinks that way :)

I'm +0.5 on adding that functionality, because then you have several scaling options that are mixed together. I can understand that it gets confusing to most users except the most advanced ones. In the waveform view, you can easily change the scaling of the waveforms with the keyboard. Idem for the probe. Add to that the independent scaling with right click and you get 3 ways to get the same thing... @nippoo wdyt? I'd be curious to have the opinion of more users (Ken, Shabnam, I-Chun, Aman, etc.)

rossant, 2015-04-29:
> I use it all the time in KV when one dimension is much larger than another, e.g. to go from this (the default original appearance):

I also agree that at least the time dimension deserves independent scaling.

rossant, 2015-04-29:
> The default behaviour should definitely be to scale the views
> independently. Perhaps holding down a modifier key could trigger rescaling
> all the views?

Yep. What about this:

* default is independent pan/zoom in the boxes
* alt + mouse wheel = same zoom gets applied to all boxes
* you cannot pan globally in all boxes together

nsteinme, 2015-04-29:
That makes good sense to me.

nippoo, 2015-04-29:
Yep, something like that is fine. Also very much +1 on the need to zoom
different dimensions independently. Sometimes you'll have a cluster with a
really elongated shape which almost looks like a straight line, and you
want to 'widen' it: normally you basically want to fill the view with
whatever cluster you're looking at, or where you want to split, or another
cluster, or whatever, and unless the cluster is circular (as opposed to
elliptical etc) you want to be able to fill it. I think...

rossant, 2015-04-30:
phew.. done!! it took me the whole day...

PR #224 (2015-05-01): Trace view
This PR implements the trace view. It also brings many other improvements and bug fixes.

* Trace view with user-settable interval and spikes with masks.
* Refactored selector: now, each view model is responsible for its selector instance. There is no more global selector instance in Session.
* A view no longer has a reference to the full spike_clusters array, just the spike clusters for the spikes shown in that view.
* Warning if a kwik file is not present.
* Improved pan zoom: in a grid view, pan zoom is now specific to each subplot (using a matrix of pan/zoom parameters with an O [... cut]
closed 2015-05-01 by nippoo