cortex-lab/phy #212: Spike sorting workflow in the notebook

rossant, 2015-04-16:
Here is an example of how a complete spike sorting workflow, from a raw `.dat` file to a fully-sorted `.kwik file`, could look like from a user perspective:

```python
# We can provide a helper CLI tool to create a standard notebook.
# >>> phy myfile.dat  # creates the notebook and open it
# >>> phy myexp.kwik --spike-detection # creates a standard notebook for spike detection
# >>> phy myexp.kwik --automatic-clustering
# >>> phy myexp.kwik --manual-clustering

# Here is an example of a standard notebook starting from a .dat file.
import phy

# The path is obtained from the CLI tool, or can be edited directly.
path = 'myfile.dat'

# Some parameters required to see the traces.
sample_rate = 20000
n_channels = 120

# This memmaps the file (fast, nothing is loaded in RAM yet).
data = phy.traces.read_dat(path)

# This displays an interactive trace view.
phy.plot.plot_traces(data, 
                     sample_rate=sample_rate, 
                     n_channels=n_channels)

# By inspecting the traces, the user generates the list of dead channels.
# They can also click on the channels to generate this list automatically.
dead_channels = [5, 85, 110]

# Next step is to load the PRB file.
# phy comes with a built-in set of common probes, but one can also
# provide their own PRB files. We can also design a tool to help the
# creation of new PRB files, and a simple PRB visualizer.
probe = phy.electrode.read_prb('128channels.prb')

# Dead channels can be set like this.
probe.dead_channels = dead_channels

# Now we create the PRM file.
params = phy.utils.create_prm('myparams.prm')
params.raw_data_files = [path]
params.kwik_file = 'mykwikfile.kwik'
params.sample_rate = sample_rate
params.probe = probe

# We launch the spike detection.
# This will automatically use multiple CPUs if
# multiple engines have been launched with IPython.parallel.
# This call is asynchronous: the user can continue to work in the notebook,
# and request the task's status.
phy.spikedetekt.run(params)

# A progress bar is then displayed.
# Reporting number of spikes detected, and estimated time remaining.

> Done in 1234 seconds! 4561232 spikes found. myfile.kwik and myfile.kwx created.

# Now we can start the automatic clustering.
phy.cluster.klustakwik.run(params)

# A progress bar is then displayed.
> Done in 2345 seconds! 789 clusters found.

# Now we can start the manual clustering.
session = phy.cluster.manual.Session()
session.open(params.kwik_file)
# ...
```

nippoo, 2015-05-28:
Related to #217

nippoo, 2015-05-28:
Related to #216

PR #366 (2015-06-24): [MRG] Spike detection
This PR brings SpikeDetekt to phy (1 CPU only for now).

* Individual spike detection processing algorithms in `phy.traces`
* Spike detection main class in `phy.cluster.algorithms`
* Improvements to the feature view
* Improved HTML view models
* Added a file logger
* Clearer color map
* Refactored feature view dimensions
* `KwikCreator` creates Kwik files.
* Load raw data from `.kwd` or `.dat` files in `KwikModel`
* Multi-shank capabilities in `MEA` class
* `session.detect_spikes()` can be used to launch spike detection on the data
* Bug fixes

### Closes

* Closes #212
* [... cut]
closed 2015-06-24 by nippoo