cortex-lab/phy #266: TraceView improvements

nippoo, 2015-05-12:
A few things to consider and prioritize:
- [ ] Jump to next spike in selected cluster(s) with Ctrl+arrow
- [ ] Show filtered data (realtime filtering with same parameters as spike detection)
- [ ] Jump to next simultaneous occurrence of spikes / all selected clusters with Ctrl+Shift+Arrow (hopefully this will be fast enough!) - within viewport aka 500ms etc
- [ ] Move backwards/forwards with mouse continuously rather than discrete jumps
- [ ] Show all spikes (not just selected clusters) by pressing S
- [ ] If selection of spikes in WaveformView is possible, jump to that spike when it's selected
- [ ] Zoom out arbitrarily to see LFP - 10 seconds? 100 seconds?
- [ ] Only show unmasked channels for cluster(s) which are selected. With 300 channels it's carnage

rossant, 2015-05-12:
* +1 to all
* except *"Move backwards/forwards with mouse continuously rather than discrete jumps"*: the benefits/cost ratio is far too low IMO
* I'd add "mode with static undersampling and no spikes to allow showing really long intervals"

rossant, 2015-06-23:
things to add:

* extra traces
* event trains (vertical lines)

closed 2019-06-13 by rossant