cortex-lab/phy #268: Name clusters

nippoo, 2015-05-12:
Naming based on natural names (clusternames.csv probably).

Should be able to toggle on/off in user parameters - default is on. Sets a `clusters/clustering_name/cluster_ID/name` attribute in the .kwik...

rossant, 2015-05-12:
in user_data maybe?

rossant, 2015-05-12:
also i guess we should have a phy-data repo for big data files like this one (like in vispy)

nippoo, 2015-05-12:
ehhhhh it's like 200kb. I don't think that's worth the extra hassle of pulling in an extra repo and symlinking in files, tbh

nippoo, 2015-05-12:
I think just name rather than user_data - consistent with channels/channel_ID/name etc.

rossant, 2015-05-12:
450kb. sorry i don't want this file in this repo..

nippoo, 2015-05-12:
well we can put it on phy.cortexlab.net/data/clusternames.csv, and automatically download it at first run of phy, but...

nippoo, 2015-05-12:
seems like a whole effort - repo size really isn't that much of a problem!

rossant, 2015-05-12:
we can do that, or do like vispy. we may very well want to add more files eventually

rossant, 2015-05-12:
yeah it is -- i want to keep repo size as low as possible

rossant, 2015-05-12:
note that we can reuse the code in utils.datasets or vispy

closed 2019-06-13 by rossant