cortex-lab/phy #271: Big Refactoring

rossant, 2015-05-13:
* New `phy.apps` package with applications (high-level API, CLI, GUI)
    * klustaviewa (Session API, Wizard GUI)
    * spikedetekt (?)
    * klustakwik (?)
    * (tempate matching gui)
* Move view model in `phy.plot.kwik.***`
* Move selector in general utils
* Keep wizard logic in phy.cluster.manual
* Refactor `phy.io`: `phy.io.mock`, `phy.io.kwik.model`, `phy.io.kwik.mock`
* Move features_masks and waveforms store items in `phy.io.kwik`
* Move session to `phy.apps.klustaviewa`

cc @nippoo

nippoo, 2015-05-13:
Where will things like a PSTH or CCG generator go? Other analysis functions which don't necessarily return graphs, but instead return numbers or a mean waveform etc etc? I'm just wondering what's best for things that could live in `phy.plot` but might not strictly be plotting-related...

rossant, 2015-05-13:
`phy.stats`?
Then we can have generic plotting functions for things like PSTH in `phy.plot`

nippoo, 2015-05-13:
Yes, that seems good. Decouple the data analysis bits from the plotting as much as possible!

PR #278 (2015-05-13): Refactoring
Closes #271
closed 2015-05-13 by nippoo