cortex-lab/phy #273: Improved similarity measure

rossant, 2015-05-13:
https://github.com/kwikteam/phy/blob/master/phy/cluster/manual/session.py#L654

@nippoo can you put here the algorithm we came up with yesterday?

rossant, 2015-05-18:
@nippoo can you remind me what was the new algo discussed over skype?

rossant, 2015-05-20:
session:346

```
            mu_0 = self.cluster_store.mean_features(target)
            mu_1 = self.cluster_store.mean_features(candidate)

            omeg_0 = self.cluster_store.mean_masks(target)
            omeg_1 = self.cluster_store.mean_masks(candidate)

            omeg_0 = np.repeat(omeg_0, self.model.n_features_per_channel)
            omeg_1 = np.repeat(omeg_1, self.model.n_features_per_channel)

            d_0 = np.dot(mu_0, omeg_0)
            d_1 = np.dot(mu_1, omeg_1)

            return (d_0 - d_1) ** 2
```

just tried: weird results. TBI...

nippoo, 2015-06-01:
Ah okay, we want:

```
   d_0 = np.dot(mu_0, omeg_0)
   d_1 = np.dot(mu_1, omeg_1)
   
   return np.linalg.norm(d_0 - d_1)
```

nippoo, 2015-06-01:
`np.multiply` is the [Hadamard product](http://en.wikipedia.org/wiki/Hadamard_product_(matrices)).

rossant, 2015-06-01:
you can also just use `mu_0 * omeg_0`

nippoo, 2015-06-01:
ugh I'd always assumed that was the dot-product. This all makes sense now...

rossant, 2015-06-01:
yeah in matlab that's the case! (i think...)

rossant, 2015-06-10:
seems to work fine with your suggestion @nippoo ! will be closed in the next PR

nsteinme, 2015-06-17:
For most clusters it looks like I get good results for closest match. But for this one cluster I had to scroll through something like 100 (not exaggerating) "closest match"es that looked like these:
[image]

[image]

before finally getting to this: 
[image]

which isn't an exact match but it's way closer than anything else. Any ideas why this would happen? The cluster doesn't look unique to me; other ones on the same channels do fine. Anyway, I don't think this is worth acting on if you don't have any ideas, since it seems to work fine for the rest of the clusters. But it can be logged here, anyway, in case we start to observe other ones in the future...

rossant, 2015-06-17:
the second cluster looks quite noisy no? in any case, yes please put the filename and cluster numbers here so that I (or @nippoo) can take a look

nsteinme, 2015-06-17:
Data in \\zserver\multichanspikes\M141003_NS2ROSS\20141112\20141112_all.kwik, cluster #17.

I don't think it is particularly noisier than any other cluster. These channels are a little raggedy-looking mostly because that section of the probe was very highly active.

rossant, 2015-06-20:
I won't have time to check this right now -- let me know if you do @nippoo

rossant, 2015-06-23:
@nsteinme is this bug critical, or can we set it to the v0.3.0 milestone?

nsteinme, 2015-06-23:
Not critical, or especially not until/unless I see it again in another
example.

nsteinme, 2015-10-04:
Ok, I would like to "re-up" this issue. I have now observed more examples of this. Somehow it prefers to give you a "most similar" cluster that has *nothing* on the channels that the "best unsorted" cluster inhabits, rather than give you ones that are on the same channels but are a bit different. It may to do with the fact that there is no normalization - so a unit with big overlapping features looks potentially "more distant" than one with *zero* overlapping features. 

Can we move this back to 0.3.0? I think it is a big problem.

rossant, 2015-10-04:
Can you try custom measures and propose one that works better for you?

nsteinme, 2015-10-04:
You think it will work? custom quality function doesn't work.
https://github.com/kwikteam/phy/issues/627

rossant, 2015-10-04:
I will have a look, in the meantime maybe you can try to change the code
directly in the source?

nsteinme, 2015-10-04:
K, I will talk with Kenneth then try something.

nsteinme, 2015-10-07:
Hey, I talked with Kenneth and we came up with some things to try. It may take a while for me to try them though because I have to find good test cases. Could you assign this issue to me, though? I can do it.

closed 2016-03-24 by rossant