cortex-lab/phy #274: Strange Attribute error when attempting to open many kwik files

shabnamkadir, 2015-05-13:
Perhaps I was attempting to open one twice.
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-7-07368081e2fb> in <module>()
      1 for human, data in spike_sorters.iteritems():
      2     print human, data['kwik_location']
----> 3     data.update({'model': KwikModel(data['kwik_location'])})
      4     #data.update({'session': Session(data['kwik_location'])})
      5 

/home/skadir/phytest/phy/phy/io/kwik_model.pyc in __init__(self, kwik_path, channel_group, clustering)
    373         self.open(kwik_path,
    374                   channel_group=channel_group,
--> 375                   clustering=clustering)
    376 
    377     # Internal properties and methods

/home/skadir/phytest/phy/phy/io/kwik_model.pyc in open(self, kwik_path, channel_group, clustering)
    648         # Open the KWIK file.
    649         self._kwik = self._open_h5_if_exists('kwik')
--> 650         if not self._kwik.is_open():
    651             raise ValueError("File {0} failed to open.".format(kwik_path))
    652         self._check_kwik_version()

AttributeError: 'NoneType' object has no attribute 'is_open'

In [5]:

spike_sorters

shabnamkadir, 2015-05-13:
OK, so jsut to let you know. The problem was caused by a typo - I pointed to a file that did not exist.

rossant, 2015-05-13:
ok I guess that a clearer message should be raised then

PR #316 (2015-06-09): KK2 integration
* [x] Create features and masks sparse structures for KK2
* [x] Make view models work without store (needed to visualize data with changing clustering)
* [x] HTML view models
* [x] Simplify handling of wizard panel in view creator
* [x] New stats view

## Closes

* Closes #221
* Closes #241
* Closes #270 
* Closes #274 
* Closes #311 
* Closes #318
* Closes #319

@thesamovar so I have written and tested a function that takes dense features and masks arrrays, and returns a `RawSparseData` structure. How should we proceed next? From a `KwikModel`, we should be able to do `run(mod [... cut]
closed 2015-06-09 by nippoo