cortex-lab/phy #291: Default features selection not useful

nippoo, 2015-05-22:
@nsteinme what do you mean by "useful"?

nsteinme, 2015-05-22:
I mean that it does not appear to pick features on which the selected
waveforms are large or well separated. As far as I can tell it picks
features on which the selected waveforms are just at background level.

rossant, 2015-05-22:
here is the current algo: https://github.com/kwikteam/phy/blob/master/phy/plot/view_models/kwik.py#L219

what do you suggest instead? for various reasons it would be slightly better to use features and masks only, not waveforms
in principle waveform amplitude should be related to feature amplitude i think...

nsteinme, 2015-05-22:
Ok, a couple things I don't understand about the algorithm but I might have an idea about why it's broken anyway. First, I don't see where you're making use of the information about which cluster(s) are selected? Second, it's not clear to me (because I don't know numpy well) whether you are taking the location of the maximum feature value on any channel anywhere - in which case you're going to pick out some feature that had one big noise blip - or if you're summing/averaging the abs of the feature value over all spikes of this cluster, in which case the following. What's probably happening is that some other feature has much larger values in general (maybe both positive and negative) than the feature you'd really like to select. So not all of the points in that other feature are large (since it's far away from the feature where this neuron actually lives and unrelated to it) but nevertheless the sum/mean of the abs comes out large. That particular issue would be solved by taking the abs after taking the sum or mean. 

If I were going to pick three features, I would pick the following: 1) The feature that best discriminates cluster#1 from the background; 2) The same for cluster#2; 3) The feature that best discriminates cluster#1 from cluster#2. "Best discriminates", in a best-possible-case scenario, would mean something like the ROC area under the curve (or any other measure of the distance between two distributions.

But a compromise sort of algorithm might be: First find those features that are unmasked on at least 50% of spikes for both clusters (if any); then find the mean amplitudes of each unit on each of those features, then take the mean of those means - *then* take the absolute value and pick the three largest. (Taking abs last solves the problem of a feature that has large positive and negative values but zero mean). 

What do you think? I'm not sure how to write it in python...

nsteinme, 2015-05-22:
and p.s. yes, the features will be related to the useful waveform features, no reason to use the waveforms themselves.

rossant, 2015-05-22:
> First, I don't see where you're making use of the information about which cluster(s) are selected?

`fet` contains the features of max 10k spikes uniformly chosen among the clusters
we could select these features differently at a small perf cost

the current algo just assigns a score to each channel: the max abs value among all spikes on that channel. there is no average so it is not resilient to outliers at all

i like your first proposition, it should be doable without too much work
@nippoo feel free to have a look at how we can implement that with numpy, scipy, or scikit-learn, or i can do it too

nsteinme, 2015-05-22:
I think you want to pick features based on the clusters selected though, definitely! Not just based on 10k spikes from all clusters. If you can use the 10k from all clusters, plus 10k (or however many there are) from each of the selected clusters, that would be optimal probably.

rossant, 2015-05-22:
sorry i meant 10k among all *selected* clusters!

nippoo, 2015-05-22:
I reckon just changing the current algorithm to use the mean (or maybe median?) highest feature could work, so we remove the outliers. No need for anything fancy?... - and follow Nick's suggestion of maximising (A vs noise, B vs noise, A vs B).

How do we handle the simultaneous selection of >2 clusters?

nsteinme, 2015-05-22:
Ok, that's better but you still need to know which came from which :) 

Looks like scikit-learn has roc area:

# Compute ROC curve and area the curve
fpr, tpr, thresholds = roc_curve(y_test, probas_[:, 1])
roc_auc = auc(fpr, tpr)
print "Area under the ROC curve : %f" % roc_auc

http://scikit-learn.org/0.11/auto_examples/plot_roc.html

But it may be best to assume they are all normally distributed, in which case the difference in means is equivalent. So if you had three variables - clust1, clust2, bckgClust - that are Nx10000 where N is the number of features, then it would be as simple as this in matlab:
mnC1 = mean(clust1,2); mnC2 = mean(clust2,2); mnBckg = mean(bckgClust,2);
feature1 = find(abs(mnC1-mnC2) == max(abs(mnC1-mnC2)), 1); 
feature2 = find(abs(mnC1-mnBckg) == max(abs(mnC1-mnBckg)), 1); 
feature3 = find(abs(mnC2-mnBckg) == max(abs(mnC2-mnBckg)), 1);

nsteinme, 2015-05-22:
For >2 clusters, if you don't plan to make the feature view expand to have more rows/cols than it currently does, I would just do the same thing as we're proposing here with the first two selected clusters. In just 4 features you simply aren't going to get all useful features if you select 10 clusters... On the other hand, you could do all pairwise differences and try to find one that maximizes them all or something like that.

nippoo, 2015-05-22:
Well, you might get all useful features: the main reason you'd select >2 is when you're comparing clusters that have - at least vaguely - similar waveforms. So actually, we could just get by taking the first 2 clusters' waveforms, and assuming the 3rd, 4th, [...] will generally work there too!

rossant, 2015-05-22:
fwiw the GUI is optimized to 2 clusters selected currently (for the wizard), although it will work with more clusters
in that case it might be sufficient to just let the user scroll through all features... or just use the first 2 by default indeed

nippoo, 2015-05-22:
Other suggestion - might it be useful to have the channels which you're comparing, boxed in green/red/blue for example, and have the featureview have similar colours? Or maybe it's not really necessary for the user to know *which* features they're comparing... as long as they discriminate enough between the clusters...

nippoo, 2015-05-22:
Also what's the best solution with the different PCs? I presume the top 3 features would sometimes come from 3 PCs from the same channel. Do we want this? We're actually picking 6 features to plot in total (x vs y), aren't we?

rossant, 2015-05-22:
a dimension = a pair (channel, feature) currently
by default, each row/col is a channel, showing PC1 nly, except in the diagonal where it is PC1/PC2

rossant, 2015-05-22:
cc @kdharris101 who might be interested in the discussion
the question is how to choose the best default projections in the feature view?

nippoo, 2015-05-22:
... so, half of the plots are identical, just rotated by 90 degrees, at the moment... A1 vs B1 <--> B1 vs A1

rossant, 2015-05-22:
yes

kdharris101, 2015-05-22:
Whatever the old KlustaViewa did was very good.

I guess the ones with largest amplitude for the selected spike?

nippoo, 2015-05-22:
The old KlustaViewa only showed a single feature...

kdharris101, 2015-05-22:
You mean a single pair of features?

rossant, 2015-05-22:
old KV: PC1 of best two channels
see https://github.com/klusta-team/klustaviewa/blob/master/klustaviewa/views/featureview.py#L922-L931
best in terms of mean feature amplitude

nippoo, 2015-05-22:
(sorry, I edited my post to say exactly that - a single pair of features rather than the 3x3 grid!). I don't know if you can see the rest of the discussion with @nsteinme 's suggestion - it's on https://github.com/kwikteam/phy/issues/291

nippoo, 2015-05-22:
@rossant I guess mean feature amplitude will be better than maximum feature amplitude and won't take long to calculate. In fact, if we cache the mean feature per cluster, we have all this information already?

rossant, 2015-05-22:
yes we already have this information so it's trivial to implement that

rossant, 2015-05-22:
OK i mark this as done (will be closed when the pr is merged). It now uses the mean feature like good ol KV
feel free to reopen if necessary

PR #288 (2015-05-26): Bug fixes
This PR brings some bug fixes, optimizations, and a significant refactoring of the cluster store. `StoreItems` now have more freedom to store and load data. There is a new `PerClusterData` that stores spike-dependent data and allows to make per-cluster selection.

* closes #283
* closes #285
* closes #290
* closes #291
* closes #292
* closes #293 
* closes #295 
* closes #296 
* closes #297 
* closes #298 
* closes #299
* closes #300
* closes #303
closed 2015-05-26 by nippoo