cortex-lab/phy #292: Some waveform pan/zoom controls missing

nippoo, 2015-05-22:
@nsteinme which particular action[s] would you need which aren't currently implemented?

nsteinme, 2015-05-22:
There's nothing that's impossible to do, but separating individual probe
sites laterally (in the x direction) could be achieved with mouse controls
in KV but is ctrl+left/right arrow in phy. It's definitely faster/nicer to
have everything on the mouse; however, like I said this is really minor...

rossant, 2015-05-22:
can you remind me what mouse controls would be effective here?

rossant, 2015-05-22:
(incidentally the reason why there are more keyboard shortcuts and less mouse shortcuts is that i now use a laptop without mouse + i have a broken hand so keyboard is easier for me :p)

rossant, 2015-05-22:
done
see https://github.com/rossant/phy/blob/bug-fixes/phy/plot/waveforms.py#L300-L313

nsteinme, 2015-05-22:
Ok, great, sorry I went back again and I can't figure out what exactly I thought was missing. I may have been confused. Apologies...

closed 2015-05-22 by nsteinme

rossant, 2015-05-22:
well I just implemented new shortcuts after your suggestions so that was definitely useful! :)

PR #288 (2015-05-26): Bug fixes
This PR brings some bug fixes, optimizations, and a significant refactoring of the cluster store. `StoreItems` now have more freedom to store and load data. There is a new `PerClusterData` that stores spike-dependent data and allows to make per-cluster selection.

* closes #283
* closes #285
* closes #290
* closes #291
* closes #292
* closes #293 
* closes #295 
* closes #296 
* closes #297 
* closes #298 
* closes #299
* closes #300
* closes #303
commit 6eae8722 (2015-05-26): Closes #292.