cortex-lab/phy #293: Crash when no KWX

nippoo, 2015-05-22:


rossant, 2015-05-22:
done: it will use default masks (1) and features (0) in this case, with a warning
note to self: in this case, the feature view might be slow on low end gpu, because there are many points overlapping at 0 with transparency and complex fragment shader (see #201). Possible fixes:

* do not show the feature view in this case
* switch to low quality features
* just display 0 or 1 spikes instead of 10k spikes at the same position

nippoo, 2015-05-22:
Uh. Can't you just plot a single point at [0, 0, 0, ..., 0] in the view? Why do you need to plot all the points?

rossant, 2015-05-22:
there is no need to! it's just the way it is architectured now... it will be easier to fix after #296

PR #288 (2015-05-26): Bug fixes
This PR brings some bug fixes, optimizations, and a significant refactoring of the cluster store. `StoreItems` now have more freedom to store and load data. There is a new `PerClusterData` that stores spike-dependent data and allows to make per-cluster selection.

* closes #283
* closes #285
* closes #290
* closes #291
* closes #292
* closes #293 
* closes #295 
* closes #296 
* closes #297 
* closes #298 
* closes #299
* closes #300
* closes #303
closed 2015-05-26 by nippoo