cortex-lab/phy #297: number of features selected for each display for each cluster should be independent

nsteinme, 2015-05-22:
Currently, by selecting 10k features total from both clusters, you get this weird effect where, when you compare one cluster with very many spikes to one with few, the one with few seems to just disappear (because indeed, a very small percentage of the 10k will belong to that cluster). It would be much better if it were 10k *per cluster*.

rossant, 2015-05-22:
i thought this was already fixed! i'll check again

rossant, 2015-05-22:
actually it's not that simple. Let's say cluster A has 9000 spikes. Cluster B has 100 spikes. Currently, since the total is less than 10k spikes, all spikes are shown, which gives the effect you describe. What would you want? Show 100 spikes from both? That would be weird I think.

nsteinme, 2015-05-22:
I left out a key phrase; what I meant to say was: "you get this weird effect where, when you compare one cluster with very many spikes to one with few, the one with few seems to just disappear *relative to when you view the one with few versus another cluster that also has few spikes*." If you like I can document some cases when a neuron has a clearly different ACG or feature view appearance when compared with cluster B than when compared with cluster C. In the case of 9000/100 - show all 9100, of course. In the case of 90000/10000 - rather than showing 9000/1000, I vote for showing 10000/10000 in featureview. And in the ACG/CCG, include all (if that is possible, or maybe up to some higher limit like 200k or something).

rossant, 2015-05-23:
yes i would like to see an example where the current behavior is wrong
it seems to me that the current selection algo does exactly what you want

nsteinme, 2015-05-23:
Ok, here is what I see when comparing 290 to 486:

[image]

And here is what I see when comparing 290 to 187:

[image]

As you can see, cluster 290's ACG appears very different in the two cases. I probably should have not tried to guess why this is happening, because maybe I'm wrong about that. But there is the bug.

rossant, 2015-05-23:
the current strategy is to take `N/n_clusters` spikes per cluster. As part of the new PR i'll change that to N, so that this number does not depend on the number of selected clusters. the drawback is a perf hit when selecting many clusters, but i think that's okay

PR #288 (2015-05-26): Bug fixes
This PR brings some bug fixes, optimizations, and a significant refactoring of the cluster store. `StoreItems` now have more freedom to store and load data. There is a new `PerClusterData` that stores spike-dependent data and allows to make per-cluster selection.

* closes #283
* closes #285
* closes #290
* closes #291
* closes #292
* closes #293 
* closes #295 
* closes #296 
* closes #297 
* closes #298 
* closes #299
* closes #300
* closes #303
closed 2015-05-26 by nippoo