cortex-lab/phy #298: autocorrelogram and crosscorrelogram should be computed for much more than 10k points (all, if possible)

nsteinme, 2015-05-22:
It's again because if you have one unit with few points relative to the other, you completely lose its ccg. Furthermore, the number of CCG refractory violations is a relevant measure and you're simply not displaying them by selecting only a subset of points. 

Could the CCGs be computed up front and cached? For ACGs can they be computed with much higher number of points included?

rossant, 2015-05-22:
i'm not sure this is a problem: the subset spikes belong to contiguous chunks
like `spikes[0:1000]`, `spikes[2000:3000]` etc where `spikes`  is the array of *all* spikes in increasing order

rossant, 2015-05-22:
btw do you know that you can change the parameters? create a `~/.phy/user_settings.py` with new parameters

nsteinme, 2015-05-22:
How long would it take to compute the CCGs for all spikes? What if there was a hotkey you could press for situations in which you really wanted to see the whole CCG (these situations are relatively rare). Or, sounds like this is a good case for the interactive - what kind of command would I have to enter to make and plot a ccg with all spikes from two clusters?

rossant, 2015-05-23:
hmm actually where did you get the 10k figure? looking at the [default parameters](https://github.com/kwikteam/phy/blob/master/phy/cluster/manual/default_settings.py#L12-L19) it looks like the CCGs and ACGs are computed from 1M spikes in 10 chunks of 100k consecutive spikes.

you can change that: create a `~/.phy/user_settings.py` and copy paste from the default settings the lines you want to change

you should also be able to add a new view with custom parameters, this should be documented at the last section of the user guide

PR #288 (2015-05-26): Bug fixes
This PR brings some bug fixes, optimizations, and a significant refactoring of the cluster store. `StoreItems` now have more freedom to store and load data. There is a new `PerClusterData` that stores spike-dependent data and allows to make per-cluster selection.

* closes #283
* closes #285
* closes #290
* closes #291
* closes #292
* closes #293 
* closes #295 
* closes #296 
* closes #297 
* closes #298 
* closes #299
* closes #300
* closes #303
closed 2015-05-26 by nippoo