cortex-lab/phy #313: Error running phy

ereming, 2015-05-28:
Hello,

I'm a brand new user (to Phy and Klusta both). I just installed Phy using the command:

`conda install conda python=3 pip numpy matplotlib scipy h5py pyqt ipython-notebook requests && pip install vispy phy`

Trying to run the software with `$ phy cluster-manual /path/file.kwik`, the data successfully loads, but then it seems like the GUI can't open. Here's the error I'm getting:

2015-05-28 13:27:28  dock:37                 PyQt is not available.
Traceback (most recent call last):
  File "/Users/Evan/miniconda/bin/phy", line 9, in <module>
    load_entry_point('phy==0.1.0', 'console_scripts', 'phy')()
  File "/Users/Evan/miniconda/lib/python3.4/site-packages/phy/scripts/phy_script.py", line 90, in main
    run(kwik_path, interactive=interactive)
  File "/Users/Evan/miniconda/lib/python3.4/site-packages/phy/scripts/phy_script.py", line 111, in run
    gui = session.show_gui(show=False)
  File "/Users/Evan/miniconda/lib/python3.4/site-packages/phy/cluster/manual/session.py", line 414, in show_gui
    gui = self.gui_creator.add(config, **kwargs)
  File "/Users/Evan/miniconda/lib/python3.4/site-packages/phy/cluster/manual/gui.py", line 403, in add
    gui = ClusterManualGUI(self.session, config=config)
  File "/Users/Evan/miniconda/lib/python3.4/site-packages/phy/cluster/manual/gui.py", line 37, in __init__
    self._dock = DockWindow(title=self.title)
  File "/Users/Evan/miniconda/lib/python3.4/site-packages/phy/utils/dock.py", line 211, in __init__
    self.setWindowTitle(title)
AttributeError: 'DockWindow' object has no attribute 'setWindowTitle'

Trying to go through the getting started tutorial, I run into a similar issue running the command `phy.enable_qt()`, getting 

2015-05-28 13:30:27  dock:37                 PyQt is not available.

I've made sure PyQt is installed, but I'm not experienced using Python, so I'm not really sure what other mistakes I could be making. I'm running on Mac OSX 10.9. Thanks for any help!

rossant, 2015-05-28:
Hi,
can you try to launch `ipython` in a terminal and then:

```
import PyQt4
import PyQt4.QtGUI
```

do you get any error message by doing that?

@nippoo can you try phy on your mac?

ereming, 2015-05-29:
Yes, here's the error. Maybe something is going wrong with the install?

In [1]: import PyQt4

In [2]: import PyQt4.QtGui
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-2-43468025f39e> in <module>()
----> 1 import PyQt4.QtGui

ImportError: dlopen(/Users/Evan/miniconda/lib/python3.4/site-packages/PyQt4/QtGui.so, 2): Library not loaded: @rpath/libpng15.15.dylib
  Referenced from: /Users/Evan/miniconda/lib/libQtGui.4.8.6.dylib
  Reason: image not found

rossant, 2015-05-29:
it looks like this bug: https://twitter.com/astrofrog/status/604039591939698688

ereming, 2015-05-29:
Ah, makes sense. Thanks! I'll try to watch to see when it gets fixed.

rossant, 2015-05-29:
can you try to execute `conda install libpng=1.5` in your terminal and try again

ereming, 2015-05-29:
That worked, thanks!

rossant, 2015-05-29:
great!

rossant, 2015-06-01:
This has now been fixed by the conda guys.

closed 2015-06-01 by rossant