cortex-lab/phy #315: IPNWLGEFPW

nippoo, 2015-05-29:
aka "ipynb with lightweight gui elements for preprocessing workflow".

Discussed during the meeting this week with @nsteinme @thesamovar @shabnamkadir .

Very related to #212. When, and to what extent, do we want to do these views? Some will be IPython widgets (parameters editor, probably for 0.3.0); some might be matplotlib/vispy WebGL/HTML GUIs (KK2/SD2 monitoring), some might be more full-fledged interactive GUI elements (R-L clustering, obviously KlustaViewa).

We should probably split this into several tasks with several milestones, but we should probably consider what individual modules we'll have. Have we ditched a central "control centre" - NDManager-esque app? 

[image]
[image]

rossant, 2015-05-29:
I would recommend to use IPython.widgets for everything and try to not use any manual HTML/JS if possible...

nsteinme, 2015-05-29:
Personally I think the IPython NB version - without a "control panel" - is
just as good as (or better than) having the full blown thing with control
panel. Further GUI elements (like a panel of buttons to run each step, or
some display summarizing the state of all the preprocessing) could probably
be added without too much trouble from there if necessary, anyway, so it
seems like a good idea to skip the control panel altogether for now.

thesamovar, 2015-05-29:
OK if Nick finds that it is all fine without a control panel and with a notebook instead, I'm happy to go with this.

nippoo, 2015-09-30:
I think we can close this for now in favour of the new pure-Python .prm with task-directed dependency graph etc? @rossant

rossant, 2015-09-30:
hmm I think that eventually, some users might still like a graphical interface (notebook-based or not), so we can probably keep this open, although with a very low priority!

nsteinme, 2015-09-30:
I think it would still be *super* useful to have the kinds of "during
preprocessing" visualization that we talked about being part of this, i.e.
see some raw data traces and choose which to exclude, see some detected
spikes and be able to choose different spikedetekt params, perhaps even see
some depiction of the clustering as it goes so you can decide whether it's
done or not. I think that stuff was the core of what we meant here.

rossant, 2015-09-30:
@nsteinme I think what you describe is related but slightly different. Having real-time visualization would indeed be useful (and relatively high priority!) and can be done without the GUI stuff this issue is about. In other words, the plan is to let you show custom and interactive visualizations in the notebook *by writing code*, rather than by being in a dedicated GUI.

closed 2016-03-24 by rossant