cortex-lab/phy #317: Remote access to phy and our data with docker and Jupyter

rossant, 2015-05-31:
* [ ] Create a docker image based on a Jupyter image with phy and vispy also installed
* [ ] In the image, mount a special `shared` folder in read-only mode (shared between the container and the host). This folder contains a list of notebooks, and symlinks to some data. There might be some tweaks to do in phy to make it work (e.g. the kwik file is currently opened in `a` mode -- that might fail on a read-only volume).
* [ ] Set up a public-facing server exposing the password-protected notebook server (running in the docker image), such that `http://.../notebooks/mynotebook.ipynb` opens directly the requested notebook
* [ ] [Upload the image on dockerhub (create a kwikteam org account if possible?)]

Now, to share a notebook, you can do this:

* Copy the notebook and data (or use symlinks) to the shared volume
* Send the notebook URL to someone
* Now, someone else can go to the URL and see and run the notebook

This is only for private sharing. For public sharing (lower priority) we can use [tmpnb](https://github.com/jupyter/tmpnb); in this case, a new temporary docker image with a fixed set of predefined notebooks is run for every new user.

rossant, 2015-06-29:
@nippoo what's the status of this? we can close, right?

nippoo, 2015-07-02:
Ah. The thought was to integrate this with Thunder / CodeNeuro right?

rossant, 2015-07-02:
i think so... closing YAGNI for now?

nippoo, 2015-07-02:
yeahhhh.

closed 2015-07-06 by rossant