cortex-lab/phy #318: add stats to wizard view

nsteinme, 2015-05-31:
Maybe I am getting too impatient about the cluster view, but would it be possible to add a list of any registered statistics to the wizard view (perhaps only if the statistic has just one element)?

So right below the cluster number and progress bar, it would just very simply say:
n_spikes = 10457
n_unmasked_channels = 23
estimated_FPrate_ISI = 0.03

(if I register a statistic that computes the estimated false positive rate from the ISI distribution, which is what I am imagining doing).

rossant, 2015-05-31:
That's a good idea, but I don't think it should belong to the wizard view. I'd rather imagine something in a dedicated view, and a bit more general. You'd basically write a function that returns an HTML string, and you'd then add this to a dedicated view in the GUI.

```python
@gui.add_view(view_type='html')
def stats_panel(cluster_ids):
    cluster = cluster_ids[0]
    return "n unmasked channels: {}".format(store.n_unmasked_channels(cluster))
```

You could also write a generic function that displays all available statistics, including your custom ones.

We could also add this stats view as one of the default views.

More generally, the idea is to let you add your very own views (VisPy views, matplotlib plots, HTML widgets, etc.). WDYT?

nsteinme, 2015-05-31:
The dedicated stats view would be fine (assuming it is connected with the
wizard to update when the selected clusters change). But I assume this will
be redundant when cluster view comes along? So not worth doing something
extremely involved, if it's just a temporary solution.

And yes, in the longer run, adding your own views is perfect - for sure I
hope I will be able to do this myself at some point!

rossant, 2015-05-31:
> assuming it is connected with the
> wizard to update when the selected clusters change

absolutely! and it will also be updated when new clusters are created following merges and splits

> But I assume this will
> be redundant when cluster view comes along?

hmm, maybe not completely redundant. We might not want put too much information in there...

rossant, 2015-06-01:
done in #316

PR #316 (2015-06-09): KK2 integration
* [x] Create features and masks sparse structures for KK2
* [x] Make view models work without store (needed to visualize data with changing clustering)
* [x] HTML view models
* [x] Simplify handling of wizard panel in view creator
* [x] New stats view

## Closes

* Closes #221
* Closes #241
* Closes #270 
* Closes #274 
* Closes #311 
* Closes #318
* Closes #319

@thesamovar so I have written and tested a function that takes dense features and masks arrrays, and returns a `RawSparseData` structure. How should we proceed next? From a `KwikModel`, we should be able to do `run(mod [... cut]
closed 2015-06-09 by nippoo