cortex-lab/phy #322: dat_channels field in PRM

nippoo, 2015-06-04:
- Ignore the channel list in the .prb.
- Have a dat_channels field (I suggest calling it that, or something similarly explicit, to avoid ambiguity).

To quote @rossant :

* every channel has a uuid as defined by the manufacturer
* this has the advantage that the position and graph will never
change: no need for a permutation
* then, in the prb or elsewhere, we have a unique "channel_mapping"
field that gives the uuid of every row in the dat file. [2, 3, -1, 0]
means that the first row is channel #2, the second is #3, the third is
dead, and the last is channel #0

Also we need to be careful about remapping the adjacency graph for dead channels to quote @nsteinme:

The only small wrinkle is that, assuming the probe's prb file comes with a connectivity graph, there will need to be a provision for adding connections - if you mark a channel as excluded, you need your connectivity graph to now change to connect across that dead site rather than using exactly the connections that are specified in the canonical prb file. [...] One way that can't miss anything, but might add more than are strictly necessary, would be: for each site that was connected to the dead site, connect it to all other sites that were connected to the dead site.

nippoo, 2015-06-04:
We should remap the channels once, and once only. Also save the remapping somewhere in the .kwik, so the TraceLoader / WaveformLoader can use this.

To investigate: is there a big performance hit in remapping? Should we instead overwrite the .dat and remap it to a trivial order before loading? If we provide a utility to remap any .dat to the correct channel order... (but maybe that's silly)

nippoo, 2015-07-02:
I reckon we should do this for 0.2.0 actually @rossant - is it a big change? Basically you're moving a field from the PRB to the PRM. Making users change their PRM files will be annoying later...

rossant, 2015-07-02:
ohhh I dunno.. add 1 week to all of our milestones

nippoo, 2015-07-02:
that long!?

rossant, 2015-07-02:
maybe I exagerate a bit, but it will change quite many places in the code, and it will require a lot of testing. But i might be wrong...

nippoo, 2015-07-02:
no, fair enough I guess - but we're already requiring everyone rewrites their PRMs now, and then changing it again in a month...

rossant, 2015-07-02:
let me put it this way: to do that properly I'll need to refactor many things about `channel_order`, and this will take quite some time. I was planning to do that for 0.3 or 0.4
so the change won't happen very soon

closed 2019-06-13 by rossant