cortex-lab/phy #324: Show statistics of multiple clusters in stats view

rossant, 2015-06-06:
Using an HTML table (row=stat, column=cluster)

rossant, 2015-06-09:
@nsteinme @nippoo do you feel this is urgently needed or not?
currently, the stats of the first selected cluster only are shown

[image]

also, this might change a bit when the cluster view is done (it won't be ready before some time)

nsteinme, 2015-06-10:
Ideally you'd want to see both, but only showing the first one should be
mostly fine. I will do some clustering with this and let you know if I'm
feeling like I really need to see both at once. Thanks!

rossant, 2015-06-10:
ok dw it's done in the PR!

closed 2015-06-16 by nippoo