cortex-lab/phy #332: Improve wizard panel

rossant, 2015-06-10:
Currently, the wizard panel is only mildly useful. The left progress bar is supposed to show you how much clusters you've seen (when 1 cluster is selected), while the right one shows you how many processed clusters you've done so far. When all clusters have been moved to a group, the bar is full.

I think we could make this panel more useful somehow, but not sure how. @nsteinme any ideas? is there any piece of information you'd want to see? We might also merge it with the stats panel if necessary...

nsteinme, 2015-06-10:
I don't know, I think it's fine! :)

rossant, 2015-06-26:
We've decided with @nsteinme and Armin to make a mini stacked bar chart showing the number of clusters in each group. Initially, it's all gray (unsorted), then it gets more colored as clusters are progressively sorted into different groups.

closed 2019-06-13 by rossant