cortex-lab/phy #334: set time as one dimension in the SingleFeatureView

nsteinme, 2015-06-10:
It used to be "ctrl-t" in KV. 

For reference, I thought I would solve this problem myself by finding the object corresponding to that feature view, and calling some method to set it to have the time feature (which I suppose is probably feature #361). Here is a record of what I tried. 

I spent a while finding gui.views[3], which is the SingleFeatureViewModel, but I can't figure out how to call its "set_dimension" or "set_dimensions" methods without a crash, if those are even the right ones to call because I can't figure out how to see documentation for them (adding ? doesn't seem to work for this, i.e. gui.views[3].set_dimensions?) and I can't find where they are defined in the code (e.g. I looked into phy/plot/features.py, but this must not be the right thing because it suggests there should be a .features method which there isn't, and what the heck is "baking"??)). I also looked into the .view and .model parts of gui.views[3] and there's a lot of stuff in there and I have no idea what's going on. How does that relate to session.model, and what is a model, anyway? 

I'm sure this will become clearer when the documentation is up to date (many of the things there don't work anymore), so I can be patient until that happens, I understand it's not top priority which is fine. But if I should be thinking about things in a different way or if I am missing some resource that will help me understand what's going on, please let me know!

rossant, 2015-06-10:
it's a bit hard to write documentation on ever-evolving code :) especially when the PR is not done yet! i'll work on the documentation just before the next release. I'll answer your specific questions tomorrow, it's getting late here ;)

nsteinme, 2015-06-10:
Yeah I'm done for tonight too. Thanks for being patient with me!

rossant, 2015-06-11:
A few definitions:

* model: an object that holds the data for your experiment
* view: a particular plot
* viewmodel: takes data from a model and creates a view

This is an instance of the [MVC](http://en.wikipedia.org/wiki/Model%E2%80%93view%E2%80%93controller) pattern: the rationale is to separate the logic for the data and the logic for the views, so you can change one independently from the rest.

`vms = gui.get_views('features')` is a list with all feature viewmodels, there's probably just one [TODO: rename this to get_view_models() instead!)

The idea is to let users deal with view_models exclusively, and not with the lower-level views. These are reserved to fairly advanced users who may want to use a different file format...

The views are in `phy.plot`. The viewmodels are currently in `phy.cluster`.

I have yet to write the doc and improve the way to specify dimensions. For now you can do 

```
vm.set_dimensions('time', (3, 0))  # x=time, y=channel 3, feature 0
```

Finally, don't worry if you all find confusing: doc will be added later. It is bad practice to write the doc at the same time you're writing the code: the code keeps changing, and the doc will be out of date soon. Generally you write your code until you're done, and before the release, you write and update the doc. You're currently working on the bleeding edge code which is not a "normal use", but again, you're far from being a "normal user" :)

rossant, 2015-06-11:
Done: pressing `t` selects time as dimension in x axis in the features view.

closed 2015-06-16 by nippoo