cortex-lab/phy #335: Features blank

lepmik, 2015-06-11:
There are no features visible 

Opening test_hybrid_120sec.kwik
Running ubuntu 14.04

rossant, 2015-06-11:
do you have a kwx and .raw.kwd files in the same folder with the same base name as the kwik?

lepmik, 2015-06-11:
yes, i downloaded the three files from (not the .zip file)

https://drive.google.com/folderview?id=0BwTrbfNJNihcaHZjTXEwdk44cFE&usp=sharing

rossant, 2015-06-11:
can you try https://github.com/kwikteam/phy-doc/blob/master/quick-start.ipynb

i'll try with these 3 files as well

rossant, 2015-06-11:
hmm works for me...

* any error message or warning in the console?
* what version of phy have you installed?
* what are your gpu and graphics drivers?

lepmik, 2015-06-11:
* No errors
 * Screendump 
[image]
 * when I select different clusters, wizard, waveform and traces seems to work fine, however traces are crammed up - opening a new issue on that
 * pulled latest version from git today
 * Drivers:
```
OpenGL vendor string: nouveau
OpenGL renderer string: Gallium 0.4 on NVD9
OpenGL core profile version string: 3.3 (Core Profile) Mesa 10.1.3
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 10.1.3
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
```
 * GPU:
```
01:00.0 VGA compatible controller: NVIDIA Corporation GF119 [NVS 310] (rev a1) (prog-if 00 [VGA controller])
	Subsystem: Hewlett-Packard Company Device 094e
	Flags: bus master, fast devsel, latency 0, IRQ 47
	Memory at f6000000 (32-bit, non-prefetchable) [size=16M]
	Memory at e8000000 (64-bit, prefetchable) [size=128M]
	Memory at f0000000 (64-bit, prefetchable) [size=32M]
	I/O ports at e000 [size=128]
	Expansion ROM at f7000000 [disabled] [size=512K]
	Capabilities: <access denied>
	Kernel driver in use: nouveau
```

rossant, 2015-06-11:
Ah, the mesa drivers are known to screw things up here. NVIDIA drivers work better. Also, how old are this computer and the GPU?

lepmik, 2015-06-11:
The computer is ca 1 yr, the gpu I dont know.

rossant, 2015-06-11:
looks like this is a GeForce 5xx(M)

do you have the possibility to install NVIDIA drivers for Ubuntu?

lepmik, 2015-06-11:
that did fix Correlations and Features :+1: However, the crammed up Traces are still there..

rossant, 2015-06-11:
good :)

crammed traces are normal and have nothing to do with the GPU: press ctrl and - to diminish the scale

closed 2015-06-16 by rossant