cortex-lab/phy #339: Cannot reopen a dataset from Python 3 phy -> Python 2 phy

nippoo, 2015-06-11:
More generally we should test for this case... this was using a clusterstore created with phy/py3, now trying to reopen on phy/py2...

```
  File "c:\users\max\development\phydev\phy\phy\io\kwik\store_items.py", line 368, in __init__
    if self.disk_store else None)
  File "c:\users\max\development\phydev\phy\phy\io\store.py", line 227, in load_file
    return cPickle.load(f)
ValueError: unsupported pickle protocol: 3
```

nippoo, 2015-06-11:
(Solution was to delete `experiment_name.phy/` folder but this should be done automatically)

rossant, 2015-06-11:
better solution would be to ensure that the store works with both versions -- I'll investigate if we can use a better format for the one file that uses pickle

nippoo, 2015-06-11:
Yes - I was about to comment - we should maybe move to protocol 2. And we should also (now we've already released phy) write code to:

- automatically un-pickle/re-pickle existing .phy dirs
- automatically regenerate the cluster store if there's a protocol error / cpickle error

> There are currently 5 different protocols which can be used for pickling. The higher the protocol used, the more recent the version of Python needed to read the pickle produced.

> - Protocol version 0 is the original “human-readable” protocol and is backwards compatible with earlier versions of Python.
> - Protocol version 1 is an old binary format which is also compatible with earlier versions of Python.
> - Protocol version 2 was introduced in Python 2.3. It provides much more efficient pickling of new-style classes. Refer to PEP 307 for information about improvements brought by protocol 2.
> - Protocol version 3 was added in Python 3.0. It has explicit support for bytes objects and cannot be unpickled by Python 2.x. This is the default protocol, and the recommended protocol when compatibility with other Python 3 versions is required.
> - Protocol version 4 was added in Python 3.4. It adds support for very large objects, pickling more kinds of objects, and some data format optimizations. Refer to PEP 3154 for information about improvements brought by protocol 4.

rossant, 2015-06-11:
ok first part done: https://github.com/rossant/phy/commit/86d84ea07df9a1d700d4128d2f5d15190442f27e
this now uses protocol 2 everywhere

next step: regenerate the store in case of error

rossant, 2015-06-11:
done

closed 2015-06-16 by nippoo