cortex-lab/phy #340: Global keyboard shortcuts (not focus-dependent)

nippoo, 2015-06-11:
We should really not have to focus a view to use its keyboard shortcuts - the entire clustering process should be doable with only a keyboard and no mouse.

We should have enough keys so as to be able to have global shortcuts, but otherwise we should be able to change focus from view to view using a single set of keys (maybe QWERTY or ASDFG or something).

On a related note, this is a good page: https://msdn.microsoft.com/en-us/library/ms971323.aspx#atg_keyboardshortcuts_keyboard_ui_design_rules

rossant, 2015-06-11:
already fixed in the latest PR, for m, o (waveforms), n (CCGs), t (time in feature view)

nippoo, 2015-06-11:
haha. maybe I should review the code before filing more issues! :P

rossant, 2015-06-11:
note that some shortcuts *are* and should be focus-dependent: like reset view, pan and zoom keyboard shortcuts, etc.

closed 2015-06-16 by nippoo