cortex-lab/phy #341: Requirements testing before install / on run

nippoo, 2015-06-11:
We should test and warn/die as necessary:

- all dependencies exist and are up-to-date (we should define and test minimum versions, linked to #115)
- GPU drivers are supported
- optionally, check latest stable version string off Web somewhere and warn if there's an update available.

We should definitely check all this when launching the cluster GUI, but maybe write some reusable code to check for any subset of this for other modules.

rossant, 2015-06-11:
Propositions:

* we build a binstar conda package that depends on specific (=latest if possible) versions of numpy h5py etc. Then dependency management is done by conda automatically.
* for GPU drivers we can try to detect if it's Mesa and throw a warning in this case

nippoo, 2015-06-11:
Yes, but people will always install from git etc, or use WinPython, or OS python, or whatever. Even if it's just people in the lab...

rossant, 2015-06-11:
hmm, but for dependencies, I'm not really aware of compatibility issues yet, so that's not that urgent.
besides, some dependencies are optional and only loaded when you need them (e.g. PyQt)
maybe we leave this particular one to the moment when we really need it?

closed 2019-06-13 by rossant