cortex-lab/phy #346: some kind of error on startup under pr345

nsteinme, 2015-06-16:
Looks like this:
[image]

I did "conda install six" which seemed to work fine, and if I go into ipython, I can do "import six" without any error. So I'm not sure if it's really a problem with six or what.

rossant, 2015-06-16:
oh this is not supposed to work. can you use `phy cluster-manual` instead?

nsteinme, 2015-06-16:
Hm, well I worked out this strategy last time I tried to phy, because "phy
is not recognized as an internal or external command". Some kind of path
issue? I am in the phy/scripts directory though.

nsteinme, 2015-06-16:
And I am in the venv that I've used phy from before.

rossant, 2015-06-16:
do `python setup.py develop` in phy's root

nsteinme, 2015-06-16:
Great, worked, thanks! Am I still using git for checking out different versions now, or has that been overwritten by this setup.py? I'm not exactly sure what it did exactly, I guess.

rossant, 2015-06-16:
yeah it's still using git. I'm not sure how it could work before actually...

closed 2015-06-16 by rossant