cortex-lab/phy #348: problem switching features with waveform view?

nsteinme, 2015-06-16:
In KV, it was the case that ctrl+left click on a channel in the waveform view made that channel's PC1 become the x-axis feature in feature view, whereas ctrl+right made it become the y-axis feature. Now, I think ctrl+left is doing the same thing, but I'm not sure what ctrl+right is doing. The problem is that if you use the "t" feature to make the x-dimension time, there is no way to see PC1 of a desired channel on the y-dimension. 

Relatedly, is there a way to choose PC2 or PC3? 

And, it would be helpful at some point to have the kind of axis labeling that KV had (nothing more complicated than a channel number and feature number on each axis, would be fine.

rossant, 2015-06-16:
* I'll fix the ctrl+click issue in waveform view
* I can add mouse/keyboard shortcuts to scroll through channels/features. I believe there are such shortcuts in KV -- do you use them? You want the same shortcuts?
* Labels are planned #164 but not for the immediate future (maybe for 0.3 or 0.4). Text happens to be extremely difficult to do in OpenGL (it's an active research problem in computer graphics). VisPy implements some of it but I have to integrate it in phy.

nippoo, 2015-06-16:
I know that @kdharris101 is very keen on replicating the keyboard/mouse shortcuts from KV as much as possible!

rossant, 2015-06-17:
good to know! I tend to prefer rethinking things from scratch over copying everything from KV, but I guess the keyboard shortcuts were good enough

nsteinme, 2015-06-17:
I don't think I used the keyboard shortcuts for scrolling through PCs much. I think the way I'd get PC2 was by ctrl+clicking a second time, and the way I'd get PC3 was just clicking the button for it in the feature view (rare).

I don't have any objection to re-thinking! :)

rossant, 2015-06-17:
OK this is fixed (in the upcoming PR), now ctrl+click and ctrl+rightclick do what you'd expect. :)

kdharris101, 2015-06-17:
Hi Cyrille,

The reason I would suggest to keep things as they were is just user inertia. (The same reason F1 still means help on windows, 30 years on.)

If things have to change, we should change them (for example 1+click, 2+click). But for anything that can stay the same, why not?

rossant, 2015-06-17:
@kdharris101 I agree that keyboard shortcuts are very hard to unlearn and change. But in the case of KV I think they were all good.

Also in the very latest version, 1+click etc have disappeared, because there is a new feature view showing a single plot (like in KV), in addition to the grid view. So I was able to bring back the old KV shortcuts (ctrl+click) instead, which work in that new feature view exclusively.

nsteinme, 2015-06-18:
I wouldn't be opposed to keeping 1+click etc for the feature grid. It
doesn't conflict with ctrl+click for singlefeatureview, does it?

rossant, 2015-06-18:
I don't mind putting it back, but will you ready need it if I implement your proposition for the dimensions in the grid view? in what situations would you want to update manually dimensions in the grid view and not in the simple feature view?

kdharris101, 2015-06-18:
Agreed, I like this feature!

PR #366 (2015-06-24): [MRG] Spike detection
This PR brings SpikeDetekt to phy (1 CPU only for now).

* Individual spike detection processing algorithms in `phy.traces`
* Spike detection main class in `phy.cluster.algorithms`
* Improvements to the feature view
* Improved HTML view models
* Added a file logger
* Clearer color map
* Refactored feature view dimensions
* `KwikCreator` creates Kwik files.
* Load raw data from `.kwd` or `.dat` files in `KwikModel`
* Multi-shank capabilities in `MEA` class
* `session.detect_spikes()` can be used to launch spike detection on the data
* Bug fixes

### Closes

* Closes #212
* [... cut]
closed 2015-06-24 by nippoo