cortex-lab/phy #349: Show detected spikes in real-time in the trace view

rossant, 2015-06-16:
The idea is to let users show raw traces at the very beginning, before any processing or file conversion (like neuroscope/kwikskope). Then, there's a little GUI where spike detection parameters can be tweaked, and detected spikes (actually, connected components in threshold crossings) are computed in real-time. This lets users find the right parameters before launching the spike detection procedure on the full dataset.

This is made possible by the modular integration of SpikeDetekt into phy (which is in good shape already). Since you're only looking at a small window at a time in the trace viewer, detecting spikes in that small window is instantaneous.

@nsteinme does that sound like something useful? Are there cases where you wished you had more feedback/interactivity when choosing SD's parameters?

nsteinme, 2015-06-17:
Yes, I think this sounds great! You would quickly be able to identify if some noisy channels needed to be eliminated or if your channel map was incorrect. 

A slight extension would be to do it on not just one segment of data from the beginning of the file, but from a few segments of data throughout the file, so that you can see whether anything went bad over time (not uncommon).

rossant, 2015-06-17:
so the way I'd see this is that spikes would be detected on whatever  chunk is displayed in the trace view
then we'll provide shortcut to jump to +10 min or whatever

nsteinme, 2015-06-17:
great, yes. Ways to navigate traceview throughout the whole file are needed
anyway...

rossant, 2015-06-17:
yep, related to #266 (cc @nippoo)

closed 2016-03-24 by rossant