cortex-lab/phy #350: training mode

nsteinme, 2015-06-17:
Was talking with @kdharris101 about an idea for how to determine whether people are doing the manual stage correctly. It's easy:

User turns on a setting to go into training/testing mode, during which a settable percentage (say 10%) of decisions they make will be about fake clusters (the rest are the real clusters they're trying to sort). So 10% of the time, instead of giving "best unsorted", it gives a fake cluster generated in one of two ways:
1) Inserting a fake cluster and splitting it. In this case, the user is supposed to correctly merge it with its other half. Could split it along peak amplitude, depth on the probe, direction of greatest variance in feature space, or time (for example). 
2) Merging two fake clusters that are not too dissimilar from each other and inserting this. The user is supposed to split it. Could parametrically vary how similar are the two clusters that are merged and should be split.

When the user gets finished and makes a decision about this cluster (in the normal way), the program tells them whether they are right or wrong, what they were supposed to do, and gives them the option to try again. If they don't try again, it removes the fake clusters and continues on as before. 

It makes a log of the tests proposed and decisions made. 

This is useful at three levels:
1) For checking that I myself [i.e. the serious user] is really doing things all right, and for giving myself a sense of how accurate I can be with this stage.
2) For checking up on technicians who might be assigned to learn and do sorting and, similarly, for training new users.
3) As a first step in the direction of developing a crowd-sourcing platform for spike sorting. If that's something one might want to do.

rossant, 2015-06-17:
I love the idea!

I'm thinking about the best way to implement it. For example, how do you create fake clusters out of an unknown dataset? You'll want the waveforms/features/CCGs to look the same for the rest of the dataset, otherwise it'll be obvious that the current cluster is the fake one.

An alternative idea that might be easier to do would be to just create one (or a couple of) test dataset with the ground truth, and enable the training mode just on these datasets for 100% of the clusters.

nsteinme, 2015-06-17:
I agree that having some ground truth full datasets is a good idea, though I'd call that a different idea. 

So how do you create fake clusters - two ways: 1) have a library of mean waveforms for each probe type and generate them from that (this would mesh with our idea of having each probe type be known with some unique identifier that specifies its geometry); 2) Implement the kind of ground truth generation that Shabnam and Kenneth have worked on, where you take a cluster from the existing dataset and shift it to a new part of the probe (for tetrodes, it would have to be from a different shank).