cortex-lab/phy #354: crash on undo

nsteinme, 2015-06-17:
I made a split in the single feature view but wanted to undo it, so I pressed ctrl+z. It gave me this error and did not appear to roll back the split.

[image]

nsteinme, 2015-06-17:
On the other hand, when I exited phy and re-opened, the clusters have the same state as they did before the original split (so maybe the undo operation did complete successfully after all).

rossant, 2015-06-17:
looking at the traceback, the undo appears to be successful, but there was a crash when computing the mean features of the clusters. Weird..

rossant, 2015-06-17:
can you double check that the `N:\\M.....` path works?

nsteinme, 2015-06-17:
Actually, now I am less sure about this - when I did the split again for the other issue I have put up - but did *not* undo it - then closed and reopened phy, the clusters again have the original state. Are things saved automatically or do I need to do a manual save before exiting?

nsteinme, 2015-06-17:
What do you mean check that the path works? I mean, it definitely did open the file successfully the first time, and any time I exit and restart the session. I could delete and rebuild the cluster store?

rossant, 2015-06-17:
you must save manually. But you should have a prompt message when quiting. How do you quit? Ctrl+Q or click on the red cross? If the latter, this is not implemented yet...

nsteinme, 2015-06-17:
ok, good to know, I didn't realize the saving behavior was different from KV.

rossant, 2015-06-17:
> What do you mean check that the path works? I mean, it definitely did open the file successfully the first time, and any time I exit and restart the session. I could delete and rebuild the cluster store?

I mean manually go to that folder in the cluster store (the path that appears in the error message) and check that the file it tried to open is there

rossant, 2015-06-17:
> ok, good to know, I didn't realize the saving behavior was different from KV.

well actually this time, it's a bug, not a feature ;)

nsteinme, 2015-06-17:
Yes, that file is there now. Let me try to reproduce the error... yes, I can reproduce it exactly, and yes the file is still there after the error. Size 2KB. Btw cluster 8 is one that shouldn't exist anymore? The two selected clusters were 8 and 11, but after I did the split, I ended up with 489, 490, and 493. Then again, doing this whole thing again - after I do the split, even though the gui now shows me 489, 490, and 493, the file 8.mean_features (and other 8. files) still exist. That's before doing undo.

rossant, 2015-06-17:
> that file is there now. Let me try to reproduce the error... yes, I can reproduce it exactly, and yes the file is still there after the error. Size 2KB.

that's correct

> Btw cluster 8 is one that shouldn't exist anymore?

yes, but the file stays here specifically in case of undo

> file 8.mean_features (and other 8. files) still exist. That's before doing undo.

ok, so that's definitely weird, I don't see why it cannot open an existing file... have you tried to close and re-open the GUI? maybe the file is locked or something... or remount the remote drive..?

nsteinme, 2015-06-17:
Closed and reopened many times now; it does always work normally on reopening.

nsteinme, 2015-06-17:
It's not a remote drive; it's a local ssd.

nsteinme, 2015-06-17:
I will delete the store.

rossant, 2015-06-17:
hmm, actually it crashed when trying to *save* the mean features, which it shouldn't do because the cluster already exist. So that's not the correct behavior -- need to fix it

rossant, 2015-06-17:
ok, can you try something? modify the function in `phy/io/kwik/store_items.py` at L345

```
    def on_cluster(self, up=None):
        super(FeatureMasks, self).on_cluster(up)
        # No need to change anything in the store if this is an undo or
        # a redo.
        if up is None or up.history is not None:
            return
        # Store the means of the new clusters.
        for cluster in up.added:
            self._store_means(cluster)
```

nsteinme, 2015-06-17:
Yes, worked! As far as I can tell everything went back to the way it was before (after also reinitializing the wizard).

rossant, 2015-06-17:
cool! just opened PR #359 fixing the bug

PR #359 (2015-06-18): [MRG] Fixed bug when computing mean features after undo
Closes #354.
closed 2015-06-18 by nippoo