cortex-lab/phy #355: split with lasso doesn't seem to work properly

nsteinme, 2015-06-17:
Here is my lasso:

[image]

And here is what I see after I press k:

[image]

rossant, 2015-06-18:
hmm, cannot reproduce this... is it systematic?

might be related to #367

nsteinme, 2015-06-22:
Here is another reproduction, now with just one cluster selected and when one of the feature axes is not time (those were two of my hypotheses):

before split:
[image]

after split:
[image]

Note orange points within the teal ones. I also see the same kind of thing when I do it in the feature grid rather than single feature view.

danieljdenman, 2015-06-22:
just want to second nick - i see this is as well. not with every every cluster, but more often than not.

rossant, 2015-06-22:
weird!! I'll have a look

rossant, 2015-06-23:
from a quick look, it seems that this happens when there are more spikes than what can be displayed in the feature view. so can you try to increase `features_n_spikes_max` and `features_grid_n_spikes_max` in your `~/.phy/user_settings.py` so that they are larger than the number of spikes in the clusters you want to split? let me know if it works. I'll work on an actual fix in the meantime...

nsteinme, 2015-06-23:
Yes! It does seem to work now that I did that. This is great as a
workaround for now - I think I wanted to have more spikes displayed,
anyway, now that I see it that way.

rossant, 2015-06-23:
fwiw i found the bug and i have almost fixed it...

danieljdenman, 2015-06-23:
i'm trying to find user_settings.py in my windows install - i see ~/.phy/internal_settings, but no user_settings.py. is there somewhere else to change the number of displayed spikes? or i can just wait for the bug fix.

rossant, 2015-06-23:
you have to create the text file manually..

nippoo, 2015-06-23:
you have to create the file yourself...

nsteinme, 2015-06-23:
You just make it and put it in ~/.phy/. Mine has literally these lines and
nothing else:

features_n_spikes_max_bg = 250000
features_n_spikes_max = 50000
features_grid_n_spikes_max = features_n_spikes_max

danieljdenman, 2015-06-23:
got it. thanks.

nippoo, 2015-06-23:
HA! Don't let anyone tell you that the tech support for phy is bad... ;-)

rossant, 2015-06-23:
@nippoo is it worth creating an *empty* file automatically if there is none? (with maybe just one line with a link to the doc or something) -- also we don't want to erase this file when the user reinstalls phy...

nippoo, 2015-06-23:
Yes, I actually wrote an issue for this but forgot to file it yesterday. We should create an empty file if there is none, perhaps ideally with a comment explaining where to find the list of params and their description/defaults

rossant, 2015-06-23:
this is now all fixed in an upcoming PR

PR #397 (2015-06-25): [MRG] Bug fixes
* Various bug fixes
* Better handling of dictionary settings
* Snippet commands

## Closes

* Closes #216
* Closes #355
* Closes #358
* Closes #375
* Closes #393
* Closes #395
closed 2015-06-25 by nippoo