cortex-lab/phy #358: higher contrast colors?

nsteinme, 2015-06-17:
Very minor point, but the default green is low contrast against the grey, e.g. I have a lot of trouble seeing which points are green vs. grey in this display: 

[image]

And given that there is no option to change cluster colors, it's something we should get right. Something with max saturation should do?

rossant, 2015-06-17:
hmm, I fear that we'll have the same problem with any color..? (you can play with the colors interactively from IPython, by tweaking the `visual.cluster_colors` -- let me know if you want more details)

what we can try is to add a white line width around the markers, I think it should greatly enhance the constrast, especially wrt background and maskes points

nsteinme, 2015-06-17:
Sorry, where do I find this visual.cluster_colors? I don't see gui.visual
or session.visual? I'm confident I can show you a better color ;)

rossant, 2015-06-17:
```
vm =  gui.get_views('features_grid')[0]
vm.view.visual.cluster_colors=[[1,0,1]]; vm.view.update()
```

rossant, 2015-06-17:
hmm you're right, this beautiful magenta is already clearer :)

the choice of colors was made relatively quickly, and i'd be happy if you could come up with a nice high-contrast colormap (we need to make sure that the colors all fit well together, I'm sure we can find some tools for that, e.g. http://paletton.com/#uid=1000u0kllllaFw0g0qFqFg0w0aF)

fyi here is the current one: https://github.com/kwikteam/phy/blob/master/phy/utils/_color.py#L46

nsteinme, 2015-06-17:
Yeah, I think it's best just to have something really bright and high saturation. This way even a single point or a thin line stands out well against black. Here were five of my favorites from KV:

[image]

nsteinme, 2015-06-17:
I'm not sure how to determine exactly whether they "fit well" - you just mean they are high contrast against each other?

rossant, 2015-06-17:
yes, but in addition to that, there is an aesthetic constraint: the colors should look nice together, basically. :)

nsteinme, 2015-06-17:
It's science, not interior decorating!!!!! ;)

rossant, 2015-06-17:
Doing science is not an excuse to make horrific software ;)

rossant, 2015-06-17:
How many colors do we need? We don't need many IMO. Currently we have 12, but most of the time only 2 are shown. For example, can you pick several high-contrast colors from http://www.materialpalette.com ?

nsteinme, 2015-06-17:
Well, how about just picking them evenly around the color wheel? This is
taking off from an idea I read once:
http://www.nature.com/nmeth/journal/v7/n8/fig_tab/nmeth0810-573_F2.html

Something similar to that but without decreasing the saturation. We have to
modify that solution since we don't know how many there will be in advance,
and you don't want a cluster's color to change as you select more.

So how about, the first two are spaced at 180 degrees around, then the next
two fill in the gaps (and have 90 degree spacing from the nearest existing
colors); the next four have 45 degree spacing, and so on. Of course they
stop being distinguishable at some point, but this is inevitable. This
might give, for instance, a bright pink and bright green to start, then add
bright orange and blue, then red, teal, yellow, and purple. That's pretty
much a list of all the best ones to work with in KV.

BTW, I do think KV is really beautiful! I think this is part of why people
use it so much! So I am not opposed to your philosophy here :)

rossant, 2015-06-17:
OK, your suggestion gives something like this:

[image]

we may change the order...

nsteinme, 2015-06-17:
Cool, looks pretty good to me though the lime and yellow are fairly close.
Also the purple is perceptually much darker than most of the rest (even if
saturation is still full). Maybe we can find some function that returns the
perceptual luminance of a given color, and skew towards picking ones that
are high on that scale?

rossant, 2015-06-17:
OK, feel free to play with it if you want.

FYI here is the colormap I used:

```
_COLORMAP = np.array([
                      [  2,217,  2],
                      [255,  2,  2],
                      [255,212,  2],
                      [ 65, 24,196],

                      [206,249,  2],
                      [189,  1,189],
                      [255,147,  2],
                      [ 14,112,188],

                      ])
```

(I used [this website](http://paletton.com/) which is pretty good)

rossant, 2015-06-18:
OK, I've updated the colormap in the PR. I think it's clearer now, but we can always tweak it further in the future.

PR #397 (2015-06-25): [MRG] Bug fixes
* Various bug fixes
* Better handling of dictionary settings
* Snippet commands

## Closes

* Closes #216
* Closes #355
* Closes #358
* Closes #375
* Closes #393
* Closes #395
closed 2015-06-25 by nippoo