cortex-lab/phy #361: Settable number of subplots in feature grid view

rossant, 2015-06-17:


rossant, 2015-06-18:
@nsteinme would that be of any use? (i.e. having3 or 5 columns/rows instead of 4)
if we go ahead with your suggestions about the dimensions in the grid view, the number of subplots will be determined by the number of features per channel

nsteinme, 2015-06-19:
Sure, although I doubt anyone would commonly want to go above 4x4 - it is already a lot! I'd say it's not a critical feature.

nsteinme, 2015-06-19:
Well, thinking about it slightly differently, if you can choose any features including custom things, then the feature grid could easily become a "scatter plot matrix" - not for spike sorting but for later analyses - in which case you would for sure want to be able to select a custom number of subplots.

closed 2016-03-24 by rossant