cortex-lab/phy #363: AssertionError in Waveforms view

danieljdenman, 2015-06-17:
this is only happening with some of my datasets, not all. not sure where the bug might be. 

runnning 'phy cluster-manual basename.kwik'. phy states normally and displays the first cluster:
[image]

when i attempt to advance to the second cluster , i get the following traceback:
[image]

and the display updates to the following for the second cluster:
[image]

the feature plots all update, as do the stats and the correlograms. the same three unmasked channels are colored, even though the waveforms have updated. however, there is some lag between the the features, etc. changing and the waveforms updating without the mask. 
this traceback also happens if i pin the cluster to compare with the best match. 
like i said, only some datasets, not others.

rossant, 2015-06-18:
could you send me your data (using dropbox gdrive or similar)? I'd need the kwik, kwx, kwd files.

danieljdenman, 2015-06-18:
sure:
https://drive.google.com/folderview?id=0B0mC-hj9DiJMfk54TE1LU1BUWUVYcDBsOFBVR2hCb05jak9COWhwUW9iU3pjOF9hLXZ3MVE&usp=sharing
[it's looking like uploading will take ~40 minutes, from now.]

anecdotally, i tried with same data again yesterday, and the got the same
error immediately upon opening, first waveform. trying again this morning,
i got as far as merging two clusters before getting the same error.

i've now reported one issue through the google group and one through
opening a github issue. do you guys have a preference? i just want to do
what's easiest for your workflow.

nippoo, 2015-06-18:
GitHub issue is much easier for us, thanks! :)

danieljdenman, 2015-06-18:
i believe i've resolved this: deleting the basename.phy folder and regenerating the cluster store fixed it.   

if i [mistakenly] run phy for the first time on the .kwik before putting the results of kk2 [in the clu] back in to the .kwik, the cluster store is generated with just the one big cluster [from spikedetekt2]. seems like if i then put the clustering in, the cluster store doesn't match the clustering. maybe a way to check for this? anyway, integrating the detection and kk2 into phy will resolve the issue.

rossant, 2015-06-18:
yes, it's a matter of days now...

rossant, 2015-06-18:
I think the solution will be to *not* create the cluster store if there is just 1 cluster -- @nippoo wdyt?

nippoo, 2015-06-18:
yes @rossant agreed - I thought we'd already decided this!

rossant, 2015-06-18:
Closing this, tracking that feature in #369

closed 2015-06-18 by rossant