cortex-lab/phy #367: mouse x coordinates in feature view off when away from origin of view

danieljdenman, 2015-06-18:
when drawing lassos, the translation of the mouse coordinates in the view into the feature space coordinates seems to depend on where in the view you click. the effect seems to depend on how far from the center of the view you are clicking.
to illustrate, i've tried to draw the same lasso around the same cluster while moving that cluster around the view.
centered in the view:
[image]

up and to the left:
[image]

down and to the right:
[image]

danieljdenman, 2015-06-18:
to clarify, the lasso translates fine if it already exists. i draw the lasso, take a screenshot, clear it, translate the feature space, and try to click the same points around the edge of the cluster, and the resulting lasso isn't quite where i clicked.

rossant, 2015-06-18:
thanks -- i can reproduce it

rossant, 2015-06-18:
a temporary work around might be to make the window as square as possible

danieljdenman, 2015-06-18:
i'll try that. it's totally workable as it is, the user just has to try a few points and use his brain to compensate...

closed 2016-03-24 by rossant