cortex-lab/phy #368: time in feature view problem

danieljdenman, 2015-06-18:
the feature vs. time plots looks weird in my data. screenshot below, and hopefully you see the same if you load the data i've sent you. from a single block recording, continuous, but all points are forced to two points, and a single point at a third timepoint. time in the traces view looks normal. 
[image]

[p.s. i really like this feature of phy. i actually asked about adding this to phy when speaking with kenneth, because i couldn't tell from my initial uses what those plots were supposed to be!]

rossant, 2015-06-18:
can you remind me where does the data come from? openephys? not spikedetekt2/klustakwik?

rossant, 2015-06-18:
i can't seem to be able to reproduce this:

[image]

am i looking at the right thing?

danieljdenman, 2015-06-18:
both - recorded with the openephys GUI into their .continuous, packed into a .dat with pack2 from here:
https://github.com/open-ephys/analysis-tools/blob/master/OpenEphys.py
and then put through spikedetekt2 and kk2.

danieljdenman, 2015-06-18:
weird. exactly like that for all my data. maybe i'll just see if it persists through the next release.

rossant, 2015-06-18:
oh, I'm running the latest PR, are you? (not sure it will change anything)

danieljdenman, 2015-06-18:
i think so. 0.2.0.dev0 installed last week.

rossant, 2015-06-18:
can you:

* run phy with `-i` (for IPython)
* `view = gui.get_views('features')[0]`
* `view.view.visual.spike_samples`

danieljdenman, 2015-06-18:
In [2]: view.view.visual.spike_samples
Out[2]: array([    1035,     9625,    17780, ..., 12217766, 12224643, 12234842],
 dtype=uint64)

rossant, 2015-06-18:
ok. python 2 or 3? Which version of Windows? 32-bit or 64-bit?

also, can you debug https://github.com/kwikteam/phy/blob/9896bc55373e9fe3e7bd0d9a69845a042716941d/phy/plot/features.py#L138 and look at the different variables? in particular look at t right before the function returns
it looks a bit like a int/float division issue...

danieljdenman, 2015-06-18:
2.7, windows7 64bit, anaconda 64bit

you got it: float division issue. changed line 136 to:
m = float(t.max())

looks good:
[image]

danieljdenman, 2015-06-18:
sorry, line 148.

rossant, 2015-06-18:
great! fixed in the latest PR

PR #392 (2015-06-23): [MRG] Fixed float division bug with time in feature view.
Closes #368
closed 2015-06-23 by rossant