cortex-lab/phy #380: h5py access error

xigumalu, 2015-06-22:
Introduced in https://github.com/kwikteam/phy/commit/262dddba0e310e0b88a30b3d3e2f2f51e7bb31f4 - on opening of any file.

```
Loading J:\Electrophys\SS026\20150203_all.kwik...
Traceback (most recent call last):
  File "c:\Program Files\KlustaViewa\WinPython-64bit-2.7.6.4\python-2.7.6.amd64\
Scripts\phy-script.py", line 9, in <module>
    load_entry_point('phy==0.2.0.dev0', 'console_scripts', 'phy')()
  File "c:\storage\workspaces\phy\phy\scripts\phy_script.py", line 90, in main
    run(kwik_path, interactive=interactive)
  File "c:\storage\workspaces\phy\phy\scripts\phy_script.py", line 106, in run
    session = Session(kwik_path)
  File "c:\storage\workspaces\phy\phy\cluster\session.py", line 72, in __init__
    gui_classes=self._gui_classes,
  File "c:\storage\workspaces\phy\phy\gui\base.py", line 605, in __init__
    self.open(path, model=model)
  File "c:\storage\workspaces\phy\phy\cluster\session.py", line 106, in open
    return super(Session, self).open(model=model, path=kwik_path)
  File "c:\storage\workspaces\phy\phy\gui\base.py", line 651, in open
    model = self._create_model(path)
  File "c:\storage\workspaces\phy\phy\cluster\session.py", line 91, in _create_m
odel
    model = KwikModel(path)
  File "c:\storage\workspaces\phy\phy\io\kwik\model.py", line 299, in __init__
    clustering=clustering)
  File "c:\storage\workspaces\phy\phy\io\kwik\model.py", line 648, in open
    self._clustering_changed(self._clustering)
  File "c:\storage\workspaces\phy\phy\io\kwik\model.py", line 723, in _clusterin
g_changed
    self._load_clustering_metadata()
  File "c:\storage\workspaces\phy\phy\io\kwik\model.py", line 513, in _load_clus
tering_metadata
    for attr in attrs}
  File "c:\storage\workspaces\phy\phy\io\kwik\model.py", line 513, in <dictcomp>

    for attr in attrs}
  File "c:\storage\workspaces\phy\phy\io\h5.py", line 165, in read_attr
    return attrs[attr_name]
  File "C:\Program Files\KlustaViewa\WinPython-64bit-2.7.6.4\python-2.7.6.amd64\
lib\site-packages\h5py\_hl\attrs.py", line 58, in __getitem__
    attr.read(arr)
  File "h5a.pyx", line 354, in h5py.h5a.AttrID.read (h5py\h5a.c:4342)
  File "_proxy.pyx", line 36, in h5py._proxy.attr_rw (h5py\_proxy.c:921)
IOError: unable to read attribute (Attribute: Read failed)
262dddba0e310e0b88a30b3d3e2f2f51e7bb31f4 is the first bad commit
commit 262dddba0e310e0b88a30b3d3e2f2f51e7bb31f4
Author: Cyrille Rossant <cyrille.rossant@gmail.com>
Date:   Wed Jun 10 18:44:21 2015 +0200

    Implemented support for clustering metadata.

:040000 040000 e60dca2ca6613f91d7fea92d165a0a4249a59da3 d11510395d2107b59226f0df
a3a0bdc533e8671e M      phy
bisect run success
```

rossant, 2015-06-22:
This should be fixed in PR #366 -- can you try it?

xigumalu, 2015-06-22:
Ahah yes, it's fixed, but there's another bug - issue #381.

PR #384 (2015-06-22): [MRG] Fix h5error
Closes #380 and #383
closed 2015-06-22 by rossant