cortex-lab/phy #381: get_html() error on phy cluster-manual

xigumalu, 2015-06-22:
```
C:\STORAGE\workspaces\phy [pr366]> phy cluster-manual J:\Electrophys\SS026\20150
203_all.kwik

Loading J:\Electrophys\SS026\20150203_all.kwik...
2015-06-22 13:58:19  h5:170                  Unable to read attribute `features_
contiguous` at `/application_data/spikedetekt/`.
2015-06-22 13:58:19  h5:170                  Unable to read attribute `save_high
` at `/application_data/spikedetekt/`.
2015-06-22 13:58:19  h5:170                  Unable to read attribute `save_low`
 at `/application_data/spikedetekt/`.
2015-06-22 13:58:19  h5:170                  Unable to read attribute `save_raw`
 at `/application_data/spikedetekt/`.
2015-06-22 13:58:19  h5:170                  Unable to read attribute `use_singl
e_threshold` at `/application_data/spikedetekt/`.
2015-06-22 13:58:19  h5:170                  Unable to read attribute `TITLE` at
 `/channel_groups/1/clusters/main`.
Features and masks initialized.←[K
Waveforms initialized.←[K
Statistics initialized.←[K9.5%.
Data successfully loaded!

Kwik file               J:\Electrophys\SS026\20150203_all.kwik
Recordings              5
List of shanks          1*
Clusterings             main*, original
Channels                120
Spikes                  8079050
Clusters                201
Duration                5461s
Traceback (most recent call last):
  File "C:\Program Files\KlustaViewa\WinPython-64bit-2.7.6.4\python-2.7.6.amd64\
Scripts\phy-script.py", line 9, in <module>
    load_entry_point('phy==0.2.0.dev0', 'console_scripts', 'phy')()
  File "c:\storage\workspaces\phy\phy\scripts\phy_script.py", line 226, in main
    exec_(cmd, globals(), locals())
  File "c:\storage\workspaces\phy\phy\ext\six.py", line 638, in exec_
    exec("""exec _code_ in _globs_, _locs_""")
  File "<string>", line 1, in <module>
  File "<string>", line 1, in <module>
  File "c:\storage\workspaces\phy\phy\scripts\phy_script.py", line 148, in run_m
anual
    gui = session.show_gui(cluster_ids=cluster_ids, show=False)
  File "c:\storage\workspaces\phy\phy\cluster\session.py", line 356, in show_gui

    **kwargs)
  File "c:\storage\workspaces\phy\phy\io\base.py", line 447, in show_gui
    **params)
  File "c:\storage\workspaces\phy\phy\gui\base.py", line 239, in add
    widget = self._create_widget(widget_class, **kwargs)
  File "c:\storage\workspaces\phy\phy\gui\base.py", line 209, in _create_widget
    return widget_class(**kwargs)
  File "c:\storage\workspaces\phy\phy\cluster\manual\gui.py", line 107, in __ini
t__
    **kwargs)
  File "c:\storage\workspaces\phy\phy\gui\base.py", line 331, in __init__
    self._initialize_views()
  File "c:\storage\workspaces\phy\phy\cluster\manual\gui.py", line 125, in _init
ialize_views
    super(ClusterManualGUI, self)._initialize_views()
  File "c:\storage\workspaces\phy\phy\gui\base.py", line 342, in _initialize_vie
ws
    requested_count=self._state.get('view_count', None),
  File "c:\storage\workspaces\phy\phy\gui\base.py", line 414, in _load_config
    self.add_view(name, **kwargs)
  File "c:\storage\workspaces\phy\phy\gui\base.py", line 459, in add_view
    item = self._view_creator.add(item, **kwargs)
  File "c:\storage\workspaces\phy\phy\gui\base.py", line 239, in add
    widget = self._create_widget(widget_class, **kwargs)
  File "c:\storage\workspaces\phy\phy\gui\base.py", line 209, in _create_widget
    return widget_class(**kwargs)
  File "c:\storage\workspaces\phy\phy\cluster\view_models.py", line 57, in __ini
t__
    **kwargs)
  File "c:\storage\workspaces\phy\phy\gui\base.py", line 42, in __init__
    self._view = self._create_view(**kwargs)
  File "c:\storage\workspaces\phy\phy\gui\base.py", line 160, in _create_view
    self._update(view, **kwargs)
  File "c:\storage\workspaces\phy\phy\gui\base.py", line 151, in _update
    html = self.get_html(**kwargs)
TypeError: get_html() got an unexpected keyword argument 'position'
```

rossant, 2015-06-22:
hmmm can you try to delete your `~/.phy/internal_settings`?

nippoo, 2015-06-22:
can confirm this is fixed if I delete internal_settings. Is this case rare enough that we should let users handle it? Or should we fix this in the code? This is a very unintuitive fix (judging by the error)

rossant, 2015-06-22:
this is a conflict between different versions of the code -- there's not much to fix IMO while this part of the code is still changing...

closed 2015-06-29 by rossant