cortex-lab/phy #383: IOerror

lepmik, 2015-06-22:
Trying to open any .kwik file
```python
IOError                                   Traceback (most recent call last)
<ipython-input-9-1329459a0d5f> in <module>()
----> 1 model = KwikModel('../test_hybrid_120sec/test_hybrid_120sec.kwik')

/home/mikkel/apps/KwikTeam/phy/phy/io/kwik/model.pyc in __init__(self, kwik_path, channel_group, clustering)
    297         self.open(kwik_path,
    298                   channel_group=channel_group,
--> 299                   clustering=clustering)
    300 
    301     @property

/home/mikkel/apps/KwikTeam/phy/phy/io/kwik/model.pyc in open(self, kwik_path, channel_group, clustering)
    652 
    653         # This loads spike clusters and cluster groups.
--> 654         self._clustering_changed(self._clustering)
    655 
    656         # This needs channels, channel_order, and waveform loader.

/home/mikkel/apps/KwikTeam/phy/phy/io/kwik/model.pyc in _clustering_changed(self, value)
    727         self._load_spike_clusters()
    728         self._load_cluster_groups()
--> 729         self._load_clustering_metadata()
    730         if _to_close:
    731             self._kwik.close()

/home/mikkel/apps/KwikTeam/phy/phy/io/kwik/model.pyc in _load_clustering_metadata(self)
    514             try:
    515                 metadata[attr] = self._kwik.read_attr(self._clustering_path,
--> 516                                                       attr)
    517             except OSError:
    518                 debug("Error when reading `{}:{}`.".format(

/home/mikkel/apps/KwikTeam/phy/phy/io/h5.pyc in read_attr(self, path, attr_name)
    164         attrs = self._h5py_file[path].attrs
    165         if attr_name in attrs:
--> 166             return attrs[attr_name]
    167         else:
    168             raise KeyError("The attribute '{0:s}'".format(attr_name) +

h5py/_objects.pyx in h5py._objects.with_phil.wrapper (-------src-dir-------/h5py-2.5.0/h5py/_objects.c:2458)()

h5py/_objects.pyx in h5py._objects.with_phil.wrapper (-------src-dir-------/h5py-2.5.0/h5py/_objects.c:2415)()

/home/mikkel/anaconda/envs/kwikteam/lib/python2.7/site-packages/h5py/_hl/attrs.pyc in __getitem__(self, name)
     71 
     72         arr = numpy.ndarray(shape, dtype=dtype, order='C')
---> 73         attr.read(arr, mtype=htype)
     74 
     75         if len(arr.shape) == 0:

h5py/_objects.pyx in h5py._objects.with_phil.wrapper (-------src-dir-------/h5py-2.5.0/h5py/_objects.c:2458)()

h5py/_objects.pyx in h5py._objects.with_phil.wrapper (-------src-dir-------/h5py-2.5.0/h5py/_objects.c:2415)()

h5py/h5a.pyx in h5py.h5a.AttrID.read (-------src-dir-------/h5py-2.5.0/h5py/h5a.c:5006)()

h5py/_proxy.pyx in h5py._proxy.attr_rw (-------src-dir-------/h5py-2.5.0/h5py/_proxy.c:897)()

IOError: Unable to read attribute (No appropriate function for conversion path)
```

rossant, 2015-06-22:
duplicate of #380 -- can you try the latest open PR?

lepmik, 2015-06-22:
Yes, now it works, however I get

```
2015-06-22 18:37:07  h5:170                  Unable to read attribute `features_contiguous` at `/application_data/spikedetekt/`.
2015-06-22 18:37:07  h5:170                  Unable to read attribute `save_high` at `/application_data/spikedetekt/`.
2015-06-22 18:37:07  h5:170                  Unable to read attribute `save_low` at `/application_data/spikedetekt/`.
2015-06-22 18:37:07  h5:170                  Unable to read attribute `save_raw` at `/application_data/spikedetekt/`.
2015-06-22 18:37:07  h5:170                  Unable to read attribute `TITLE` at `/channel_groups/0/clusters/main`.
```

rossant, 2015-06-22:
yeah, that's normal -- these messages should probably be hidden actually since they are not useful to the user most of the time. @nippoo wdyt?

rossant, 2015-06-22:
this is now hidden (in the PR). closing

closed 2015-06-22 by rossant