cortex-lab/phy #385: KeyError

lepmik, 2015-06-22:
When I try to open a .kwik file generated by open ephys I get

```python
KeyError                                  Traceback (most recent call last)
<ipython-input-3-2e7450094fad> in <module>()
----> 1 model = KwikModel('experiment0.kwik')

/home/mikkel/apps/KwikTeam/phy/phy/io/kwik/model.py in __init__(self, kwik_path, channel_group, clustering)
    297         self.open(kwik_path,
    298                   channel_group=channel_group,
--> 299                   clustering=clustering)
    300 
    301     @property

/home/mikkel/apps/KwikTeam/phy/phy/io/kwik/model.py in open(self, kwik_path, channel_group, clustering)
    632 
    633         # Load the data.
--> 634         self._load_meta()
    635 
    636         # This needs metadata.

/home/mikkel/apps/KwikTeam/phy/phy/io/kwik/model.py in _load_meta(self)
    402         # Automatically load all metadata from spikedetekt group.
    403         path = '/application_data/spikedetekt/'
--> 404         metadata_fields = self._kwik.attrs(path)
    405         for field in metadata_fields:
    406             if field.islower() and not field.startswith('_'):

/home/mikkel/apps/KwikTeam/phy/phy/io/h5.py in attrs(self, path)
    191     def attrs(self, path='/'):
    192         """Return the list of attributes at the given path."""
--> 193         return sorted(self._h5py_file[path].attrs)
    194 
    195     def has_attr(self, path, attr_name):

h5py/_objects.pyx in h5py._objects.with_phil.wrapper (-------src-dir-------/h5py-2.5.0/h5py/_objects.c:2458)()

h5py/_objects.pyx in h5py._objects.with_phil.wrapper (-------src-dir-------/h5py-2.5.0/h5py/_objects.c:2415)()

/home/mikkel/anaconda/envs/kwikteam/lib/python2.7/site-packages/h5py/_hl/group.pyc in __getitem__(self, name)
    162                 raise ValueError("Invalid HDF5 object reference")
    163         else:
--> 164             oid = h5o.open(self.id, self._e(name), lapl=self._lapl)
    165 
    166         otype = h5i.get_type(oid)

h5py/_objects.pyx in h5py._objects.with_phil.wrapper (-------src-dir-------/h5py-2.5.0/h5py/_objects.c:2458)()

h5py/_objects.pyx in h5py._objects.with_phil.wrapper (-------src-dir-------/h5py-2.5.0/h5py/_objects.c:2415)()

h5py/h5o.pyx in h5py.h5o.open (-------src-dir-------/h5py-2.5.0/h5py/h5o.c:3507)()

KeyError: 'Unable to open object (Component not found)'
```

rossant, 2015-06-22:
can you send me the data?

lepmik, 2015-06-22:
Yes, the raw file was originally named experiment0_100.raw.kwd, just renamed it.

https://www.dropbox.com/sh/5u7e3e4n1jja2uu/AAAlBkdfhnQvFWzgxliCozjaa?dl=0

rossant, 2015-06-22:
Note to self: the bug comes from the kwik file being basically empty -- the model should instead just load the raw data (from `.raw.kwd`). The sample rate can also be found there. So the fix will be to detect this and just load the raw data in the model.

rossant, 2015-06-23:
superseded by #391

closed 2015-06-23 by rossant