cortex-lab/phy #394: Keyboard shortcut "snippets" in GUI

nippoo, 2015-06-23:
So this would be a flexible, easy-to-use, powerful interface to change parameters without having dedicated keyboard shortcuts. Inspired by vim's keyboard shortcuts...

Examples:
- change correlogram width (`:cv w 100` displays 100ms width) or bin size (`:cv bin 10` = 10ms bin size) or both (`cv bin 10 w 100`) or number of spikes `cv spikes 1000`
- skip to traceview time (`tv 1000` skips to 1000 seconds or `tv 1000 1001` shows that time duration or something)
- select clusters (`c 1004` or `c 1004 23 430 2101` or `c 1000 to 1023` or `c Nick Alex Cyrille Bob Alice Kenneth` or something)
- select featureview features (`fv 9.1 10.3 10.1 t` selects channel 9 feature 1, channel 10 feature 3, channel 10 feature 1, time)
- maybe waveforms/features visible `fv bg 100000` or `fv fg 10000` (featureview foreground/background 100000) or `fv bg default` or `wv fg 1000` (waveformview foreground 1000 spikes) 

to quote @rossant:
* something like `:c 10 20<enter>` shows those 2 clusters. when you press `:`, you're in "snippet mode", and everything you type goes to the status bar. if you type backspace, it erases a character as expected. press enter to execute the snippet and clear the status bar. also, press `esc` to leave the snippet mode.
* the functionality would be built into BaseGUI, which ClusterManualGUI derives from.
* the list of snippets is completely extensible, and the default ones could be part of `default_settings.py`. a snippet is defined by a key-value pair where the key is a regex, and the value is a function that takes as argument the gui and the regex output
* example

```python

def select_clusters(gui, r):
    cluster = r.group(1)
    gui.select(cluster)

cluster_manual_snippets = {
  r'c ([\d]+)': select_clusters,
  # ...
}
```

* you can also add snippets interactively:

```python
@gui.register_snippet(r'c ([\d]+)')
def select_clusters(gui, r):
    pass
```

thoughts @nsteinme ?

nsteinme, 2015-06-23:
Sounds very useful! Great concept. I think I would probably use all of the
examples you give. If it's as easy to implement as you make it look,
definitely go for it.

rossant, 2015-06-23:
this is basically done :)
was relatively straightforward...
here is an example of how it works:

 ```python
def _select_clusters(gui, args):
    args = list(map(int, args.split(' ')))
    gui.select(args)

cluster_manual_snippets =  {
    'c': _select_clusters,
}
```

(no need for regexes)

rossant, 2015-06-24:
you can now `:c 5 10` so select two clusters, or `:c 5-10` to select 5 clusters...

PR #418 (2015-06-29): Improve phy command-line tool and spike detection
This PR improves the `phy` script and fixes several bugs.

* Use subparsers in `phy` CLI tool
* `phy traces` CLI script to visualize raw traces
* Show cluster group in stats panel
* Add library of probe files
* Bug fixes

## Closes

* Closes #214 
* Closes #378 
* Closes #394 
* Closes #403 
* Closes #404 
* Closes #412 
* Closes #413 
* Closes #416
* Closes #419
* Closes #421
* Closes #430
* Closes #432
* Closes #435
* Closes #439 
* Closes #441
closed 2015-06-29 by nippoo