cortex-lab/phy #398: on mac, view settings don't persist between sessions?

nsteinme, 2015-06-24:
On windows, if I change around the positions of the views in cluster-manual then close phy, they will keep those new positions when I reopen (great!). This doesn't happen on mac, perhaps related to the fact that it doesn't seem to close properly when I press command+q? (I have to go to the ipython window and type "exit").

rossant, 2015-06-24:
* The views are not saved if you close a view (due to a Qt bug). Can you reproduce the problem if you don't close any view?
* Can you quit the program by clicking on the window manager "close" button (red cross or whatever)
* Do you have the same exit problem without IPython (ie without `-i` option)
* Can @nippoo reproduce the problem on his mac?

rossant, 2015-06-24:
* Can you reproduce the problem in debug mode (`-d` option) and copy-paste the console output?
* What is the Python setup on the Mac?

nsteinme, 2015-06-24:
Hm, it seems to be working now - not sure why not before? Sometimes I get a segfault when exiting, not sure how to reproduce that repeatedly, but maybe that was what happened when it lost my view settings.

rossant, 2015-06-29:
I also see the segfault when exiting on linux from time to time -- opening a new issue now

closed 2015-06-29 by rossant