cortex-lab/phy #399: Split by amplitude

nippoo, 2015-06-24:
This function will take a while to run. Can be launched from a snippet - ideally click on a point in WaveformView then launch snippet with ```:wv split-amplitude```

Function prototype along the lines of:

```
def split_by_amplitude(cluster, amplitude, channel, sample_offset):
# get waveforms
waveforms = session.model.waveforms[(session.model.spikes_clusters == cluster)]
newcluster1 = waveforms(channel,sample_offset,:) >= amplitude)
newcluster2 = waveforms(channel,sample_offset,:) < amplitude)
```

Will be slow but only needs to load waveforms from 1 cluster!

rossant, 2015-06-24:
It may be worth putting *all* waveforms in the cluster store. They are actually already stored now during spike detection, all of them, so why not reuse them in the GUI! The user could have an option to not save them if they want to save space, or they could just delete the folder.

We could store them in one block instead of cluster per cluster -- this will be much faster than now, and the performance of the manual clustering GUI won't be impacte;

nippoo, 2015-06-25:
I don't know if this will be worth the space hit... I don't know if this feature will be used often. The waveforms could potentially be huge; even stored in a sparse format... Why would you think it's worth putting them into the cluster store?

nippoo, 2015-06-25:
(Also I love the idea we're re-inventing the Klusters format in flat binaries)...

rossant, 2015-06-25:
> Why would you think it's worth putting them into the cluster store?

maybe because they're actually **already** computed when using the new spike detection algorithm... so if they're already here, why not use them?

rossant, 2015-06-25:
> (Also I love the idea we're re-inventing the Klusters format in flat binaries)...

as Ken says, hopefully we're not going around in circles by in spirals...

nippoo, 2015-06-25:
Right. But the main thing here is saving disk space - this is why we don't store them! ATM in dense format we're effectively just recreating the KWX. In a sparse array on disk it's still going to be sizeable. The *only* reason we'll use them is to do this amplitude-splitting, right?

I suggest we wait and see what the performance impact is. It might be good to have it as an option but probably off-by-default...

rossant, 2015-06-30:
So what's the status of this? Anything urgent? Will this still be a problem once the waveforms are subsampled regularly from the data? @nsteinme @nippoo

nsteinme, 2015-06-30:
The regular subsampling may have been the explanation for the particular example I showed. But it doesn't change the fact that this is a feature people want to see. It also would have quickly told me that I was looking at an artifact rather than me having to think of it myself a week later ;) 

I don't suppose it was ever urgent, but I think it should still be a planned feature.

rossant, 2019-06-13:
split by amplitude is possible in the dev branch

closed 2019-06-13 by rossant