cortex-lab/phy #401: autoscale button for features/waveforms views

nsteinme, 2015-06-24:
To make the [99% of?] data fill the viewable area.

rossant, 2015-06-24:
* can use `p = np.percentile(self.view.visual.features, 95)`
* requires #402

nsteinme, 2015-10-04:
The more I sort, the more I realize how critical this is. Basically, there is still a 30-second gap between making one decision and being in a position to make another, just like with KV - the difference is, it doesn't feel so bad now because it's all spent with me zooming out, finding the waveform, zooming in on it, ctrl+clicking to pick some features - i.e. because I'm doing something active rather than just sitting and waiting. But it is still a major time sink.

rossant, 2015-10-04:
Yes we will try to do this this week @nippoo

rossant, 2015-10-04:
There should not be a button for this: the view should *automatically* zoom in, which will also remove a lot of complexity in the code due to saving the manually-set scaling factor.

nsteinme, 2015-10-04:
Yeah, good, let me be more clear about what I'm hoping will happen. Upon assigning the current "best unsorted" to some cluster_group, then: whether by hotkey or automatically, either fine for me, you go from this:

[waveform view of where the last best unsorted used to be]
[image]

to this: [waveform view of where the new one is, with same zoom/scaling parameters?]
[image]

and from this [feature view of old features]
[image]

to this [feature view of new features, keeping time as the x-axis if it was before, or else picking two large features for "best unsorted"]
[image]

When you go from one "closest match" to another, nothing should change in either view. 

Of course it will also help if this one is also fixed: https://github.com/kwikteam/phy/issues/617

There is another problem this will generate - at least when you have to zoom in and out all the time, you do keep track of *where* on the probe you are currently looking. This is relevant, important information. The coolest and nicest solution might be if it did a really quick google-maps style animation of zooming out to show the whole probe then zooming in to the new spot. A simpler workaround could be opening a second waveform view (is this possible?) that the user keeps zoomed out so they see both the zoomed and un-zoomed view at once. A third option would be some kind of labeling of the waveform view with geometrical coordinates.

rossant, 2015-10-04:
wow you're a tough user to satisfy ;)

How urgent is all of this? I'm currently doing a significant refactoring of the code, and I could tackle some of this at the same time, but it won't be ready until after SfN. Otherwise we can try to hack something quicker, but we'll have to rework the implementation on the new version.

nsteinme, 2015-10-04:
Can we say 0.3? If it's not ready before SfN that's fine. I understand the
refactoring is important, no worries. And don't worry, I am mostly very
satisfied, these are just some loose ends. Thanks!

rossant, 2015-10-04:
Oh yeah, sure, actually I've put the "Pre-SfN" milestone but it'll probably have to be relabelled ;)

Pierre and Olivier have also requested this feature for template matching, which is understandable given the size of their probes...

nippoo, 2015-10-04:
No, this is definitely definitely high-priority (at least most of it). It annoys me even when I'm trying to look through a dataset quickly, never mind sorting a whole one by hand...!

Related to issue #94.

closed 2016-03-24 by rossant